chipsec/chipsec #37: unsupported plateforme

darcosion, 2016-06-21:
ERROR: Unsupported Platform: VID = 0x8086, DID = 0x0044
ERROR: Platform is not supported (Unsupported Platform: VID = 0x8086, DID = 0x0044).

Do you want more information to complete your tool ?

c7zero, 2016-07-01:
DID 0x0044 is old Core mobile (Westmere) processor with Ibexpeak PCH. Chipsec doesn't support anything older than Sandy Bridge. You can run chipsec with "-i" command-line option. Functionality which doesn't depend on the platform will run correctly but platform dependent functionality will probably be incorrect. Alternatively, add this line to Chipset_Dictionary in chipset.py so that chipsec can recognize this system:
0x0044 : {'name' : 'Westmere',       'id' : 107, 'code' : 'WSM',  'longname' : 'Westmere' }

darcosion, 2016-07-01:
I'm not sure this is my processor, look that : 
`$ sudo dmidecode -t 4
# dmidecode 2.12
SMBIOS 2.7 present.

Handle 0x001F, DMI type 4, 42 bytes
Processor Information
	Socket Designation: U3E1
	Type: Central Processor
	Family: Core i3
	Manufacturer: Intel(R) Corporation
	ID: 55 06 02 00 FF FB EB BF
	Signature: Type 0, Family 6, Model 37, Stepping 5
	Flags:
		FPU (Floating-point unit on-chip)
		VME (Virtual mode extension)
		DE (Debugging extension)
		PSE (Page size extension)
		TSC (Time stamp counter)
		MSR (Model specific registers)
		PAE (Physical address extension)
		MCE (Machine check exception)
		CX8 (CMPXCHG8 instruction supported)
		APIC (On-chip APIC hardware supported)
		SEP (Fast system call)
		MTRR (Memory type range registers)
		PGE (Page global enable)
		MCA (Machine check architecture)
		CMOV (Conditional move instruction supported)
		PAT (Page attribute table)
		PSE-36 (36-bit page size extension)
		CLFSH (CLFLUSH instruction supported)
		DS (Debug store)
		ACPI (ACPI supported)
		MMX (MMX technology supported)
		FXSR (FXSAVE and FXSTOR instructions supported)
		SSE (Streaming SIMD extensions)
		SSE2 (Streaming SIMD extensions 2)
		SS (Self-snoop)
		HTT (Multi-threading)
		TM (Thermal monitor supported)
		PBE (Pending break enabled)
	Version: Intel(R) Core(TM) i3 CPU       M 370  @ 2.40GHz
	Voltage: 0.0 V
	External Clock: Unknown
	Max Speed: 4000 MHz
	Current Speed: 2409 MHz
	Status: Populated, Enabled
	Upgrade: ZIF Socket
	L1 Cache Handle: 0x0023
	L2 Cache Handle: 0x0022
	L3 Cache Handle: 0x0020
	Serial Number: Not Specified
	Asset Tag: FFFF
	Part Number: Not Specified
	Core Count: 2
	Core Enabled: 2
	Thread Count: 4
	Characteristics:
		64-bit capable
`
When i start chipsec, it's work but they are many modules who didn't work : 
`[CHIPSEC] ***************************  SUMMARY  ***************************
[CHIPSEC] Time elapsed          0.009
[CHIPSEC] Modules total         14
[CHIPSEC] Modules failed to run 2:
ERROR: chipsec.modules.common.smrr
ERROR: chipsec.modules.smm_dma
[CHIPSEC] Modules passed        3:
[+] PASSED: chipsec.modules.common.bios_ts
[+] PASSED: chipsec.modules.common.spi_desc
[+] PASSED: chipsec.modules.common.bios_kbrd_buffer
[CHIPSEC] Modules failed        3:
[-] FAILED: chipsec.modules.common.bios_wp
[-] FAILED: chipsec.modules.common.bios_smi
[-] FAILED: chipsec.modules.common.spi_lock
[CHIPSEC] Modules with warnings 0:
[CHIPSEC] Modules skipped 6:
[*] SKIPPED: chipsec.modules.common.smm
[*] SKIPPED: chipsec.modules.common.uefi.access_uefispec
[*] SKIPPED: chipsec.modules.common.uefi.s3bootscript
[*] SKIPPED: chipsec.modules.common.secureboot.variables
[*] SKIPPED: chipsec.modules.module_template
[*] SKIPPED: chipsec.modules.remap
[CHIPSEC] Modules with Exceptions 2:
ERROR: chipsec.modules.common.smrr
ERROR: chipsec.modules.smm_dma
[CHIPSEC] *****************************************************************
`

And some modules have right errors : ERROR: Exception occurred during chipsec.modules.smm_dma.run(): '/home/darcosion/chipsec/source/tool/chipsec/helper/linux/cores.so: cannot open shared object file: No such file or directory'

I have added the line into Chipset Dictionnary and that, it's work but i'm not sure if it's the good solution.

c7zero, 2016-10-27:
There are two issues here:
1. cores.so file needs to be built (please see chipsec-manual.pdf how to do that)
2. It's the first generation Core M processor (as I wrote above) not supported by chipsec. You can use generic functionality (chipsec_util.py) but most modules which depend on the platform/chipset won't work properly even if you add a line in chipset.py with DID 0x44

I'm closing the issue.

closed 2016-10-27 by c7zero