chipsec/chipsec #39: install and run for the first time

liad344, 2016-06-29:
when writing "sc create chipsec binpath=<PATH_TO_CHIPSEC_SYS> type= kernel DisplayName="Chipsec driver" as the manual says to do to run chipsec
on cmd , it says "access is denied" why? and what does it mean?

bryongloden, 2016-06-29:
Greetings @liad344. How are you

liad344, 2016-06-29:
I’m fine , and you@bryongloden  ?

bryongloden, 2016-06-29:
Yeah <3. I'm alright

liad344, 2016-06-30:
and about the problem? ..anyone?

tweksteen, 2016-06-30:
Are you running the command as Administrator? The current code does automatically create and start the service if required. Have you tried to run chipsec_main.py directly?

liad344, 2016-06-30:
Hi ,i do run this aa administrator , ill try to run chipsec_main.py
directly

c7zero, 2016-07-01:
Note that you need to reboot Windows with driver signature enforcement turned off. Check the manual on how to do that

c7zero, 2016-08-09:
Closing the issue

closed 2016-08-09 by c7zero