chipsec/chipsec #60: module organization

chipsecintel, 2016-08-12:
We've been thinking about reorganizing the location of modules. Currently, we have some under common and some directly in the modules directory. There's not any real difference. 

It might make sense to create a new structure to organize modules by type (eg. spi, smm, uefi, etc.) instead. 

Issue is to discuss and make changes (or just close if we decide not to change the organization).

tweksteen, 2016-08-14:
It sounds like a good idea. Any chance to reconsider the naming of "modules" to something a bit less generic, such as "checks"?

c7zero, 2016-08-15:
Because modules can be anything (checks, testing tools, PoCs) I'd like to keep generic "modules".

I'll add proposed change to the module organization however I'd like to keep flexibility to organize modules by architecture/vendor/platform and by technology.

BrentHoltsclaw, 2019-12-06:
Closing issue.  Please see #747.

closed 2019-12-06 by BrentHoltsclaw