chipsec/chipsec #61: cores.so does not get compiled when you compile the linux driver

chipsecintel, 2016-08-12:
We intended to have the linux driver Makefile call the cores.so makefile. However, this is not done.

closed 2016-08-13 by c7zero

tweksteen, 2016-08-14:
This should not be. The driver and the linux helper native code should not be coupled. It seems that it is currently done as an easy "installation" step where one can simply compile the driver and run the python code. 

The reality is that the setup.py will already take care of compiling the extension with the correct settings (see the Extension class. The Makefile in the linux helper directory should not be used). Ideally, the setup.py could include building the kernel module as an extra command. One could then chain the instruction to build the kernel module and run the tests or build the module and start a develop mode:

    python setup.py build build_driver tests or
    python setup.py build build_driver develop

without having to rely on external shell script or similar.

chipsecintel, 2016-08-15:
There is no instance in which you should build the driver and not make the ring3 software. There may be instances where you build the ring3 software but not build the driver. Therefore, the driver makefile should call the ring3 makefile. Until/unless we completely deprecate this, it is definitely a bug to add an undocumented step to do the build from two different places. 

Regarding setup.py, it's probably best to open a separate issue regarding the changes needed in setup.py. I'd rather have it build the driver by default and pass something like "no_driver" to have it not do so. That'd be consistent with the rest of the usage.