chipsec/chipsec #81: Parse EFI variables problem with NVAR and EFI_FIRMWARE_FILE_SYSTEM2_GUID

Sebastienbr, 2016-09-02:
Hi,

I am trying to parse EFI variables with this command but it does not work. 

 >chipsec_util.py decode rom.bin nvar

No EFI variables are dumped except for the DEFAULT_NVAR.

After looking into chipsec source, I think my problem is related to this part of the code:
https://github.com/chipsec/chipsec/blob/48e3a3b58e7daf1cb45480f0d735c693f28a6320/source/tool/chipsec/hal/spi_uefi.py#L457

My rom FV format is EFI_FIRMWARE_FILE_SYSTEM2_GUID and the variables are stored in NVAR_NVRAM_FS_FILE.

The code below seems to solve my problem. Do you think it's a correct solution to implement? 

```
elif fwbin.Type == EFI_FV_FILETYPE_RAW:
	if fwbin.Name == NVAR_NVRAM_FS_FILE: 
		if fwbin.UD:
			if dumpall: _uefi.parse_EFI_variables( os.path.join(fwbin_dir, 'DEFAULT_NVRAM'), fv.Image, False, FWType.EFI_FW_TYPE_NVAR )
		else:
			if dumpall: _uefi.parse_EFI_variables( os.path.join(fwbin_dir, 'NVRAM'), fv.Image, False, FWType.EFI_FW_TYPE_NVAR )
```

Also, I have notice that this part of code was still using the FvImage variable so I have change it for fv.Image.

Thanks,

Regards,

Sebastien

chipsecintel, 2016-09-02:
Hi Sebastien,

 

This piece of code is particularly for DEFAULT_NVRAM, you should check rom.bin.dir/nvram_nvar.nvram.dir folder for regular variables.

 

Thank you,

Andrew.

 

https://github.com/chipsec/chipsec/blob/48e3a3b58e7daf1cb45480f0d735c693f28a6320/source/tool/chipsec/hal/spi_uefi.py#L457

Sebastienbr, 2016-09-02:
Thank you Andrew for the reply. 

I am trying to diff 2 dumps to compare setup variables (similar to exercise 4.5 in your training). 

Not sure why, but the values in rom.bin.dir/nvram_nvar.nvram.dir does not change when I diff this folder. 

However, when I use the patch below, I can see the value changed when I diff the 00_8C8CE578-8A3D-4F1C-9935-896185C32DD3 folder.

Thanks,

Sebastien

chipsecintel, 2016-09-02:
Hi Sebastien,

 

It’s hard to say why this occurs. Do you mind sharing the images you are parsing?

 

Thank you,

Andrew.

andrew-furtak, 2016-09-06:
It appeared that getNVstore_NVAR returned wrong NV store.  This way you always have the store with UD flag in both locations. I corrected the getNVstore_NVAR routine and added changes you proposed.

Sebastienbr, 2016-09-06:
Works well, thank you!

closed 2016-09-07 by c7zero