chipsec/chipsec #90: help- need to save a list of set parameters

liad344, 2016-09-18:
hi, 
I'm new to chipsec.
chipsec is running on UEFI shell through usb thumb drive. i want it to save a list of set parameters about the computer- lets say - amount of available RAM, number of CPU cores, pcie lanes and so on..  

how can I do it? thanks

liad344, 2016-09-19:
??

johnloucaides, 2017-03-15:
There are a lot of things that it can save, either by manually reading registers using chipsec_util or from logging results of chipsec_main (identifying the platform generation and various register state in addition to the pass/fail results). However, we don't specifically have available RAM, CPU cores, or PCIe lanes. You might try checking the output of chipec_util.py pci list, though.

c7zero, 2017-03-19:
These also may help but chipsec isn't generic tool to get system info. You may want to look at wmic and systeminfo utilities on Windows.

These may also help:
chipsec_util cpu info
chipsec_util cpu cpuid

Closing this issue.

closed 2017-03-19 by c7zero