chipsec/chipsec #104: Can't build Win7 driver with x86 enviroment command prompt

hcwang1126, 2016-10-13:
At least can't build the driver well on my working machine: Win10, DDK: 7600.16385.1.

The root cause is that the enviroment is specific for WinXP and function KeQueryActiveProcessorCount is not available. Using  x64 command prompt can build the driver without any error.

The document readme might need to update.

andrew-furtak, 2017-03-14:
My Win10 shows WinXP environment as the only x86 environment available. You need to use environment from Start Menu\Programs\Windows Driver Kits\WDK 7600.16385.1\Build Environments\Windows 7. I'll update the readme file with correct build environment.

c7zero, 2017-03-16:
Does this resolve the issue?

hcwang1126, 2017-03-16:
I think so, thanks. -HC

________________________________
寄件者: c7zero <notifications@github.com>
寄件日期: 2017年3月16日 上午 08:22:54
收件者: chipsec/chipsec
副本: hcwang1126; Author
主旨: Re: [chipsec/chipsec] Can't build Win7 driver with x86 enviroment command prompt (#104)

Does this resolve the issue?

closed 2017-03-16 by c7zero