chipsec/chipsec #126: recompile driver on kernel upgrade for linux distro packaging

johnloucaides, 2016-11-19:
For linux distros that package chipsec, we will want them to recompile the driver whenever they update the kernel. This may mean that we should store the driver in a separate place from the code. This issue is to figure out the path forward and resolve.

Some options for now:
 * pass a path for where to put the driver as an option to setup.py
 * call driver makefile each time you run chipsec to ensure it is up to date
 * leave as is (linux distro packages may break when distro updates kernel)

archey, 2016-11-28:
It might be a good idea to remove the building of kernel module from the build scripts, this would allow pkgers to include it as a seperate pkg or a dependency of the chipsec tool. It could be updated as the kernel updates in each distro, to allow it to run accordingly.

tweksteen, 2016-11-29:
It is already possible to build the package without building the driver: `python setup.py install --skip-driver` or `pip install --install-option="--skip-driver" chipsec`

archey, 2016-12-02:
The skip-driver part is fine, its just figuring out a good place to drop the driver to allow distros to build it. I Believe the concensus is going to be drop it into dkms to rebuild it and have it be loaded and unloaded as needed by the tool.

c7zero, 2017-01-23:
With dkms.conf and --skip-driver install option, I believe we have this issue solved. I intend to close unless there are objections.

closed 2017-03-08 by chipsecintel