chipsec/chipsec #140: can't build with GCC plugins( e.g: latent_entropy)

citypw, 2017-01-18:
Platform: Debian 8( x86_64) + kernel 4.9

> root@debian8-test:/home/shawn/chipsec# python setup.py install
running install
running bdist_egg
running egg_info
writing chipsec.egg-info/PKG-INFO
writing top-level names to chipsec.egg-info/top_level.txt
writing dependency_links to chipsec.egg-info/dependency_links.txt
writing entry points to chipsec.egg-info/entry_points.txt
reading manifest file 'chipsec.egg-info/SOURCES.txt'
writing manifest file 'chipsec.egg-info/SOURCES.txt'
installing library code to build/bdist.linux-x86_64/egg
running install_lib
running build_py
creating build
creating build/lib.linux-x86_64-2.7
copying chipsec_main.py -> build/lib.linux-x86_64-2.7
copying chipsec_util.py -> build/lib.linux-x86_64-2.7
creating build/lib.linux-x86_64-2.7/chipsec_tools
copying chipsec_tools/__init__.py -> build/lib.linux-x86_64-2.7/chipsec_tools
creating build/lib.linux-x86_64-2.7/chipsec
copying chipsec/module_common.py -> build/lib.linux-x86_64-2.7/chipsec
copying chipsec/chipset.py -> build/lib.linux-x86_64-2.7/chipsec
copying chipsec/logger.py -> build/lib.linux-x86_64-2.7/chipsec
copying chipsec/module.py -> build/lib.linux-x86_64-2.7/chipsec
copying chipsec/file.py -> build/lib.linux-x86_64-2.7/chipsec
copying chipsec/__init__.py -> build/lib.linux-x86_64-2.7/chipsec
copying chipsec/defines.py -> build/lib.linux-x86_64-2.7/chipsec
copying chipsec/command.py -> build/lib.linux-x86_64-2.7/chipsec
copying chipsec/xmlout.py -> build/lib.linux-x86_64-2.7/chipsec
creating build/lib.linux-x86_64-2.7/chipsec_tools/linux
copying chipsec_tools/linux/__init__.py -> build/lib.linux-x86_64-2.7/chipsec_tools/linux
creating build/lib.linux-x86_64-2.7/chipsec_tools/windows
copying chipsec_tools/windows/__init__.py -> build/lib.linux-x86_64-2.7/chipsec_tools/windows
creating build/lib.linux-x86_64-2.7/chipsec/helper
copying chipsec/helper/helpers.py -> build/lib.linux-x86_64-2.7/chipsec/helper
copying chipsec/helper/oshelper.py -> build/lib.linux-x86_64-2.7/chipsec/helper
copying chipsec/helper/__init__.py -> build/lib.linux-x86_64-2.7/chipsec/helper
creating build/lib.linux-x86_64-2.7/chipsec/hal
copying chipsec/hal/uefi_common.py -> build/lib.linux-x86_64-2.7/chipsec/hal
copying chipsec/hal/cmos.py -> build/lib.linux-x86_64-2.7/chipsec/hal
copying chipsec/hal/io.py -> build/lib.linux-x86_64-2.7/chipsec/hal
copying chipsec/hal/paging.py -> build/lib.linux-x86_64-2.7/chipsec/hal
copying chipsec/hal/cpuid.py -> build/lib.linux-x86_64-2.7/chipsec/hal
copying chipsec/hal/mmio.py -> build/lib.linux-x86_64-2.7/chipsec/hal
copying chipsec/hal/msr.py -> build/lib.linux-x86_64-2.7/chipsec/hal
copying chipsec/hal/tpm12_commands.py -> build/lib.linux-x86_64-2.7/chipsec/hal
copying chipsec/hal/iommu.py -> build/lib.linux-x86_64-2.7/chipsec/hal
copying chipsec/hal/smbus.py -> build/lib.linux-x86_64-2.7/chipsec/hal
copying chipsec/hal/pci.py -> build/lib.linux-x86_64-2.7/chipsec/hal
copying chipsec/hal/interrupts.py -> build/lib.linux-x86_64-2.7/chipsec/hal
copying chipsec/hal/uefi_search.py -> build/lib.linux-x86_64-2.7/chipsec/hal
copying chipsec/hal/spi_uefi.py -> build/lib.linux-x86_64-2.7/chipsec/hal
copying chipsec/hal/spi.py -> build/lib.linux-x86_64-2.7/chipsec/hal
copying chipsec/hal/ec.py -> build/lib.linux-x86_64-2.7/chipsec/hal
copying chipsec/hal/acpi.py -> build/lib.linux-x86_64-2.7/chipsec/hal
copying chipsec/hal/msgbus.py -> build/lib.linux-x86_64-2.7/chipsec/hal
copying chipsec/hal/ucode.py -> build/lib.linux-x86_64-2.7/chipsec/hal
copying chipsec/hal/uefi.py -> build/lib.linux-x86_64-2.7/chipsec/hal
copying chipsec/hal/spd.py -> build/lib.linux-x86_64-2.7/chipsec/hal
copying chipsec/hal/spi_descriptor.py -> build/lib.linux-x86_64-2.7/chipsec/hal
copying chipsec/hal/vmm.py -> build/lib.linux-x86_64-2.7/chipsec/hal
copying chipsec/hal/__init__.py -> build/lib.linux-x86_64-2.7/chipsec/hal
copying chipsec/hal/cpu.py -> build/lib.linux-x86_64-2.7/chipsec/hal
copying chipsec/hal/physmem.py -> build/lib.linux-x86_64-2.7/chipsec/hal
copying chipsec/hal/uefi_platform.py -> build/lib.linux-x86_64-2.7/chipsec/hal
copying chipsec/hal/hal_base.py -> build/lib.linux-x86_64-2.7/chipsec/hal
copying chipsec/hal/iobar.py -> build/lib.linux-x86_64-2.7/chipsec/hal
copying chipsec/hal/pcidb.py -> build/lib.linux-x86_64-2.7/chipsec/hal
[... 187 lines cut ...]
  File "/usr/lib/python2.7/distutils/dist.py", line 972, in run_command
    cmd_obj.run()
  File "/usr/lib/python2.7/dist-packages/setuptools/command/bdist_egg.py", line 161, in run
    cmd = self.call_command('install_lib', warn_dir=0)
  File "/usr/lib/python2.7/dist-packages/setuptools/command/bdist_egg.py", line 147, in call_command
    self.run_command(cmdname)
  File "/usr/lib/python2.7/distutils/cmd.py", line 326, in run_command
    self.distribution.run_command(command)
  File "/usr/lib/python2.7/distutils/dist.py", line 972, in run_command
    cmd_obj.run()
  File "/usr/lib/python2.7/dist-packages/setuptools/command/install_lib.py", line 22, in run
    self.build()
  File "/usr/lib/python2.7/distutils/command/install_lib.py", line 111, in build
    self.run_command('build_ext')
  File "/usr/lib/python2.7/distutils/cmd.py", line 326, in run_command
    self.distribution.run_command(command)
  File "/usr/lib/python2.7/distutils/dist.py", line 972, in run_command
    cmd_obj.run()
  File "setup.py", line 112, in run
    self._build_linux_driver()
  File "setup.py", line 77, in _build_linux_driver
    subprocess.check_output(["make", "-C", build_driver])
  File "/usr/lib/python2.7/subprocess.py", line 573, in check_output
    raise CalledProcessError(retcode, cmd, output=output)
subprocess.CalledProcessError: Command '['make', '-C', '/home/shawn/chipsec/build/lib.linux-x86_64-2.7/drivers/linux']' returned non-zero exit status 2

tweksteen, 2017-01-18:
There seems to be an issue with your GCC plugin itself: 

    cc1: error: cannot load plugin ./scripts/gcc-plugins/latent_entropy_plugin.so
    ./scripts/gcc-plugins/latent_entropy_plugin.so: undefined symbol: symtab

Does it build fine without the plugins enabled?

citypw, 2017-01-19:
Yes, it's working perfect fine without GCC plugins.

tweksteen, 2017-01-20:
I'm closing this issue then. Feel free to reopen if you think this issue is related to Chipsec and not your plugin.

closed 2017-01-20 by tweksteen

citypw, 2017-01-20:
Hi, this plugin is not "mine" technically;-) Some GNU/Linux distro might ship with some GCC plugins in the future. The latent_entropy was merged( ported from PaX/Grsecurity) in v4.9. Chipsec will face this problem soon or later.

tweksteen, 2017-01-22:
I agree that we should support this plugin if this is part of the official kernel. However, the error you reported seems to be related to the plugin (or how the plugin is installed) and not to Chipsec itself.
./scripts/gcc-plugins/latent_entropy_plugin.so: undefined symbol: symtab

citypw, 2017-01-23:
I built & installed kernel v4.9 in the default way. CONFIG_GCC_PLUGIN_LATENT_ENTROPY & CONFIG_GCC_PLUGINS are set in the build config and that's all.