chipsec/chipsec #143: chipsec_util not working on macOS

hydrajump, 2017-01-23:
`chipsec_main` works fine, but when I run `chipsec_util.py uefi <any command>` I get the following error:

`ERROR: 'OSXHelper' object has no attribute 'driver_loaded'`

I've tried both using chipsec via pip and installing manually.

macOS 10.12.2
chipsec 1.2.5

tweksteen, 2017-01-23:
Thanks for the report, I'll look into it.

c7zero, 2017-01-23:
I believe, OSXHelper is missing the following in its __init__:
        super(OSXHelper, self).__init__()

hydrajump, 2017-01-24:
@c7zero that seems to work to get past that error. However, it produces this:

```
ERROR: Could not load helper for 'Darwin' environment (unsupported environment?)
Traceback (most recent call last):
  File "chipsec_util.py", line 43, in <module>
    _cs = cs()
  File "/Users/test/Downloads/chipsec-1.2.5/chipsec/chipset.py", line 696, in cs
    if _chipset is None: _chipset = Chipset( helper() )
  File "/Users/test/Downloads/chipsec-1.2.5/chipsec/helper/oshelper.py", line 502, in helper
    _helper  = OsHelper()
  File "/Users/test/Downloads/chipsec-1.2.5/chipsec/helper/oshelper.py", line 108, in __init__
    raise OsHelperError( "Could not load helper for '%s' environment (unsupported environment?)" % os_system, errno.ENODEV )
chipsec.helper.oshelper.OsHelperError: Could not load helper for 'Darwin' environment (unsupported environment?)
```

tweksteen, 2017-01-27:
For now, the EFI variables related commands are not yet implemented on OSX.

With @c7zero's fix, I have another error returned to me: `TypeError: argument must be an int, or have a fileno() method.` In this case, the OSX driver does not have the ability to read the PCI configuration register of the CPU to determine its codename. If forcing to use the driver (modifying `requires_driver` in `uefi_cmd.py`), it stills misses the implementation of `list_EFI_variables`.

But I wasn't able to reproduce the second error you are mentioning. Was it with using pip? If so, my guess is that the OSX helper/driver is not correctly packaged. Otherwise, could you please reproduce using this repository master branch?

I'll create some feature requests for what I've just described. @hydrajump feel free to send patches to implement these :)

c7zero, 2017-03-16:
@tweksteen this new error should be fixed with PR #181

c7zero, 2017-03-19:
chipsec_util.py works on macOS now

Some functionality isn't available on macOS. This is tracked by issue #149. 
Closing this issue.

closed 2017-03-19 by c7zero