chipsec/chipsec #166: OS X 10.11.6: errors with chipsec_util.py

JayBrown, 2017-03-11:
OS X 10.11.6, mid-2010 MBP6,2, SMC v1.58f17, BootROM MBP61.0057.B11

Building the kext with Xcode was fine. Loading it was apparently too… only had to set it to root:wheel beforehand:

```
sudo kextutil ~/GitHub/chipsec/chipsec/helper/osx/chipsec.kext 
Password:
Diagnostics for ~/GitHub/chipsec/chipsec/helper/osx/chipsec.kext:
Code Signing Failure: not code signed
kext-dev-mode allowing invalid signature -67062 0xFFFFFFFFFFFEFA0A for kext "~/GitHub/chipsec/chipsec/helper/osx/chipsec.kext"
kext signature failure override allowing invalid signature -67062 0xFFFFFFFFFFFEFA0A for kext "~/GitHub/chipsec/chipsec/helper/osx/chipsec.kext"
```

Then, with the python scripts, the errors began:

```
sudo python chipsec_util.py spi info
Password:

################################################################
##                                                            ##
##  CHIPSEC: Platform Hardware Security Assessment Framework  ##
##                                                            ##
################################################################
[CHIPSEC] Version 1.2.5
[CHIPSEC] API mode: using CHIPSEC kernel module API
ERROR: Unsupported Platform: VID = 0x8086, DID = 0x0044
WARNING: *******************************************************************
WARNING: * Unknown platform!
WARNING: * Platform dependent functionality will likely be incorrect
WARNING: * Error Message: "Unsupported Platform: VID = 0x8086, DID = 0x0044"
WARNING: *******************************************************************
[CHIPSEC] Executing command 'spi' with args ['info']

Traceback (most recent call last):
  File "chipsec_util.py", line 263, in <module>
    sys.exit( main() )
  File "chipsec_util.py", line 259, in main
    return chipsecUtil.main()
  File "chipsec_util.py", line 226, in main
    comm.run()
  File "~/GitHub/chipsec/chipsec/utilcmd/spi_cmd.py", line 69, in run
    _spi = SPI( self.cs )
  File "/~/GitHub/chipsec/chipsec/hal/spi.py", line 174, in __init__
    self.cs.helper.map_io_space(self.rcba_spi_base, Cfg.SPI_MMIO_BASE_LENGTH, 0)
  File "~/GitHub/chipsec/chipsec/helper/oshelper.py", line 250, in map_io_space
    return self.helper.map_io_space(physical_address, length, cache_type)
AttributeError: 'OSXHelper' object has no attribute 'map_io_space'
```

…and…

```
sudo python chipsec_util.py spi dump rom.bin                                                                                  [12:41:41]
Password:

################################################################
##                                                            ##
##  CHIPSEC: Platform Hardware Security Assessment Framework  ##
##                                                            ##
################################################################
[CHIPSEC] Version 1.2.5
[CHIPSEC] API mode: using CHIPSEC kernel module API
ERROR: Unsupported Platform: VID = 0x8086, DID = 0x0044
WARNING: *******************************************************************
WARNING: * Unknown platform!
WARNING: * Platform dependent functionality will likely be incorrect
WARNING: * Error Message: "Unsupported Platform: VID = 0x8086, DID = 0x0044"
WARNING: *******************************************************************
[CHIPSEC] Executing command 'spi' with args ['dump', 'rom.bin']

Traceback (most recent call last):
  File "chipsec_util.py", line 263, in <module>
    sys.exit( main() )
  File "chipsec_util.py", line 259, in main
    return chipsecUtil.main()
  File "chipsec_util.py", line 226, in main
    comm.run()
  File "~/GitHub/chipsec/chipsec/utilcmd/spi_cmd.py", line 69, in run
    _spi = SPI( self.cs )
  File "~/GitHub/chipsec/chipsec/hal/spi.py", line 174, in __init__
    self.cs.helper.map_io_space(self.rcba_spi_base, Cfg.SPI_MMIO_BASE_LENGTH, 0)
  File "~/GitHub/chipsec/chipsec/helper/oshelper.py", line 250, in map_io_space
    return self.helper.map_io_space(physical_address, length, cache_type)
AttributeError: 'OSXHelper' object has no attribute 'map_io_space'
```

c7zero, 2017-03-11:
But root caused. Thanks! Fixing.. Temporarily, you can replace "except oshelper.UnimplementedAPIError:" with "except:" in line 175 in chipsec/hal/spi.py

c7zero, 2017-03-11:
err "Bug"

c7zero, 2017-03-11:
should be fixed with pr #167

JayBrown, 2017-03-11:
Thank you - worked; got the rom.bin dump and the spi info results… just don't know what to make of it / do with it. LOL.

c7zero, 2017-03-12:
you can now deconstruct it and extract all executables and configuration. Use: "python chipsec_util.py decode rom.bin"

JayBrown, 2017-03-12:
Yes, I already figured that much, but it doesn't work. I'm always getting

```
[CHIPSEC] API mode: using OS native API (not using CHIPSEC kernel module)
ERROR: argument must be an int, or have a fileno() method.
```

(The kext is loaded; checked with `kextstat | grep -v com.apple`.)

I looked in the manual and tried e.g. the option `vss` at the end, or `spi decode rom.bin`, but no difference.

JayBrown, 2017-03-12:
Also tried the following method (blacklist check), but received an error:

```
python chipsec_main.py -i --no_driver -m tools.uefi.blacklist -a rom.bin,blacklist.json                                       [08:38:41]
[*] Ignoring unsupported platform warning and continue execution
################################################################
##                                                            ##
##  CHIPSEC: Platform Hardware Security Assessment Framework  ##
##                                                            ##
################################################################
[CHIPSEC] Version 1.2.5
[CHIPSEC] Arguments: -i --no_driver -m tools.uefi.blacklist -a rom.bin,blacklist.json
[CHIPSEC] API mode: using OS native API (not using CHIPSEC kernel module)
[-] Traceback (most recent call last):
  File "chipsec_main.py", line 567, in main
    self._cs.init( self._platform, (not self._no_driver), self._driver_exists )
  File "~/GitHub/chipsec/chipsec/chipset.py", line 287, in init
    vid_did  = self.pci.read_dword( 0, 0, 0, 0 )
  File "~/GitHub/chipsec/chipsec/hal/pci.py", line 249, in read_dword
    value = self.helper.read_pci_reg( bus, device, function, address, 4 )
  File "~/GitHub/chipsec/chipsec/helper/oshelper.py", line 185, in read_pci_reg
    return self.helper.read_pci_reg( bus, device, function, address, size )
  File "~/GitHub/chipsec/chipsec/helper/osx/helper.py", line 138, in read_pci_reg
    ret = self.ioctl(IOCTL_RDPCI, data)
  File "~/GitHub/chipsec/chipsec/helper/osx/helper.py", line 113, in ioctl
    return fcntl.ioctl(self.dev_fh, ioctl_name, args)
TypeError: argument must be an int, or have a fileno() method.
```

c7zero, 2017-03-13:
The error is because of --no_driver option is specified so ioctl cannot find the device. The real issue is chipsec tries to detect the platform regardless of --no_driver option... I couldn't figure this out today.
Dropping --no_driver option will get you through this bug but will fail when trying to decompress the executables out of the image. Will be looking into this.

Unfortunately, Mac OS support is in alpha stage, so a lot is not implemented yet. Easier is to boot Mac off of USB with Linux and extract and parse firmware from there where compression tools are compiled when installing chipsec and driver is loaded automatically.

c7zero, 2017-03-16:
Should be fixed with PR #181

JayBrown, 2017-03-16:
Great! Will check it out tonight.

JayBrown, 2017-03-16:
Doesn't work yet. Running any command gets me:

```
Traceback (most recent call last):
  File "chipsec_util.py", line 39, in <module>
    from chipsec.chipset import UnknownChipsetError
  File "~/chipsec/chipsec/chipset.py", line 43, in <module>
    from chipsec.helper.oshelper import OsHelper, OsHelperError
  File "~/chipsec/chipsec/helper/oshelper.py", line 93, in <module>
    import chipsec.helper.helpers
  File "~/chipsec/chipsec/helper/helpers.py", line 24, in <module>
    from chipsec.helper.osx import *
  File "~/chipsec/chipsec/helper/osx/helper.py", line 30, in <module>
    from chipsec_tools import efi_compressor
ImportError: cannot import name efi_compressor
```

chipsecintel, 2017-03-16:
Did you setup.py?

It should support Mac OS X efi_compressor.

Like: python setup.py build_ext –I

JayBrown, 2017-03-17:
I hadn't, but after your message I just did. Still getting the same error messages. The setup.py, however, seems to have worked:

```
 ❯ python setup.py build_ext -I $HOME/chipsec
running build_ext
building 'chipsec_tools.efi_compressor' extension
creating build
creating build/temp.macosx-10.11-intel-2.7
creating build/temp.macosx-10.11-intel-2.7/chipsec_tools
creating build/temp.macosx-10.11-intel-2.7/chipsec_tools/compression
creating build/temp.macosx-10.11-intel-2.7/chipsec_tools/compression/LZMA
creating build/temp.macosx-10.11-intel-2.7/chipsec_tools/compression/LZMA/SDK
creating build/temp.macosx-10.11-intel-2.7/chipsec_tools/compression/LZMA/SDK/C
creating build/temp.macosx-10.11-intel-2.7/chipsec_tools/compression/Tiano
cc -fno-strict-aliasing -fno-common -dynamic -arch i386 -arch x86_64 -g -Os -pipe -fno-common -fno-strict-aliasing -fwrapv -DENABLE_DTRACE -DMACOSX -DNDEBUG -Wall -Wstrict-prototypes -Wshorten-64-to-32 -DNDEBUG -g -fwrapv -Os -Wall -Wstrict-prototypes -DENABLE_DTRACE -arch i386 -arch x86_64 -pipe -Ichipsec_tools/compression/Include -I~/chipsec -I/System/Library/Frameworks/Python.framework/Versions/2.7/include/python2.7 -c chipsec_tools/compression/EfiCompressor.c -o build/temp.macosx-10.11-intel-2.7/chipsec_tools/compression/EfiCompressor.o
cc -fno-strict-aliasing -fno-common -dynamic -arch i386 -arch x86_64 -g -Os -pipe -fno-common -fno-strict-aliasing -fwrapv -DENABLE_DTRACE -DMACOSX -DNDEBUG -Wall -Wstrict-prototypes -Wshorten-64-to-32 -DNDEBUG -g -fwrapv -Os -Wall -Wstrict-prototypes -DENABLE_DTRACE -arch i386 -arch x86_64 -pipe -Ichipsec_tools/compression/Include -I~/chipsec -I/System/Library/Frameworks/Python.framework/Versions/2.7/include/python2.7 -c chipsec_tools/compression/LZMA/LzmaCompress.c -o build/temp.macosx-10.11-intel-2.7/chipsec_tools/compression/LZMA/LzmaCompress.o
cc -fno-strict-aliasing -fno-common -dynamic -arch i386 -arch x86_64 -g -Os -pipe -fno-common -fno-strict-aliasing -fwrapv -DENABLE_DTRACE -DMACOSX -DNDEBUG -Wall -Wstrict-prototypes -Wshorten-64-to-32 -DNDEBUG -g -fwrapv -Os -Wall -Wstrict-prototypes -DENABLE_DTRACE -arch i386 -arch x86_64 -pipe -Ichipsec_tools/compression/Include -I~/chipsec -I/System/Library/Frameworks/Python.framework/Versions/2.7/include/python2.7 -c chipsec_tools/compression/LZMA/LzmaDecompress.c -o build/temp.macosx-10.11-intel-2.7/chipsec_tools/compression/LZMA/LzmaDecompress.o
cc -fno-strict-aliasing -fno-common -dynamic -arch i386 -arch x86_64 -g -Os -pipe -fno-common -fno-strict-aliasing -fwrapv -DENABLE_DTRACE -DMACOSX -DNDEBUG -Wall -Wstrict-prototypes -Wshorten-64-to-32 -DNDEBUG -g -fwrapv -Os -Wall -Wstrict-prototypes -DENABLE_DTRACE -arch i386 -arch x86_64 -pipe -Ichipsec_tools/compression/Include -I~/chipsec -I/System/Library/Frameworks/Python.framework/Versions/2.7/include/python2.7 -c chipsec_tools/compression/LZMA/SDK/C/Bra86.c -o build/temp.macosx-10.11-intel-2.7/chipsec_tools/compression/LZMA/SDK/C/Bra86.o
cc -fno-strict-aliasing -fno-common -dynamic -arch i386 -arch x86_64 -g -Os -pipe -fno-common -fno-strict-aliasing -fwrapv -DENABLE_DTRACE -DMACOSX -DNDEBUG -Wall -Wstrict-prototypes -Wshorten-64-to-32 -DNDEBUG -g -fwrapv -Os -Wall -Wstrict-prototypes -DENABLE_DTRACE -arch i386 -arch x86_64 -pipe -Ichipsec_tools/compression/Include -I~/chipsec -I/System/Library/Frameworks/Python.framework/Versions/2.7/include/python2.7 -c chipsec_tools/compression/LZMA/SDK/C/LzFind.c -o build/temp.macosx-10.11-intel-2.7/chipsec_tools/compression/LZMA/SDK/C/LzFind.o
cc -fno-strict-aliasing -fno-common -dynamic -arch i386 -arch x86_64 -g -Os -pipe -fno-common -fno-strict-aliasing -fwrapv -DENABLE_DTRACE -DMACOSX -DNDEBUG -Wall -Wstrict-prototypes -Wshorten-64-to-32 -DNDEBUG -g -fwrapv -Os -Wall -Wstrict-prototypes -DENABLE_DTRACE -arch i386 -arch x86_64 -pipe -Ichipsec_tools/compression/Include -I~/chipsec -I/System/Library/Frameworks/Python.framework/Versions/2.7/include/python2.7 -c chipsec_tools/compression/LZMA/SDK/C/LzmaDec.c -o build/temp.macosx-10.11-intel-2.7/chipsec_tools/compression/LZMA/SDK/C/LzmaDec.o
cc -fno-strict-aliasing -fno-common -dynamic -arch i386 -arch x86_64 -g -Os -pipe -fno-common -fno-strict-aliasing -fwrapv -DENABLE_DTRACE -DMACOSX -DNDEBUG -Wall -Wstrict-prototypes -Wshorten-64-to-32 -DNDEBUG -g -fwrapv -Os -Wall -Wstrict-prototypes -DENABLE_DTRACE -arch i386 -arch x86_64 -pipe -Ichipsec_tools/compression/Include -I~/chipsec -I/System/Library/Frameworks/Python.framework/Versions/2.7/include/python2.7 -c chipsec_tools/compression/LZMA/SDK/C/LzmaEnc.c -o build/temp.macosx-10.11-intel-2.7/chipsec_tools/compression/LZMA/SDK/C/LzmaEnc.o
cc -fno-strict-aliasing -fno-common -dynamic -arch i386 -arch x86_64 -g -Os -pipe -fno-common -fno-strict-aliasing -fwrapv -DENABLE_DTRACE -DMACOSX -DNDEBUG -Wall -Wstrict-prototypes -Wshorten-64-to-32 -DNDEBUG -g -fwrapv -Os -Wall -Wstrict-prototypes -DENABLE_DTRACE -arch i386 -arch x86_64 -pipe -Ichipsec_tools/compression/Include -I~/chipsec -I/System/Library/Frameworks/Python.framework/Versions/2.7/include/python2.7 -c chipsec_tools/compression/Tiano/Decompress.c -o build/temp.macosx-10.11-intel-2.7/chipsec_tools/compression/Tiano/Decompress.o
cc -fno-strict-aliasing -fno-common -dynamic -arch i386 -arch x86_64 -g -Os -pipe -fno-common -fno-strict-aliasing -fwrapv -DENABLE_DTRACE -DMACOSX -DNDEBUG -Wall -Wstrict-prototypes -Wshorten-64-to-32 -DNDEBUG -g -fwrapv -Os -Wall -Wstrict-prototypes -DENABLE_DTRACE -arch i386 -arch x86_64 -pipe -Ichipsec_tools/compression [... cut ...]
copying ~/chipsec/build/lib.macosx-10.11-intel-2.7/drivers/osx/build/Release/chipsec.kext/Contents/Info.plist -> ~/chipsec/build/lib.macosx-10.11-intel-2.7/chipsec/helper/osx/chipsec.kext/Contents
creating ~/chipsec/build/lib.macosx-10.11-intel-2.7/chipsec/helper/osx/chipsec.kext/Contents/MacOS
copying ~/chipsec/build/lib.macosx-10.11-intel-2.7/drivers/osx/build/Release/chipsec.kext/Contents/MacOS/chipsec -> ~/chipsec/build/lib.macosx-10.11-intel-2.7/chipsec/helper/osx/chipsec.kext/Contents/MacOS
removing '~/chipsec/build/lib.macosx-10.11-intel-2.7/drivers' (and everything under it)
```

JayBrown, 2017-03-17:
Not sure though if this was necessary in the first place because I already built the kext with Xcode, and already did a successful firmware dump. (?)

JayBrown, 2017-03-17:
Addendum: the kext in the build folder doesn't have the right ownership, so maybe the setup.py build script should be run with sudo, in order to automatically set the kext to root:wheel (recursive). Otherwise you can load the kext with kextutil.

Sae86, 2018-01-26:
Hi, I was getting ImportError: cannot import name efi_compressor as well but it did go away when running the above commands:
python setup.py build_ext -i
sudo python chipsec_main.py

BrentHoltsclaw, 2019-01-05:
Unable to reproduce MacOS build problems with current release (MacOS 10.14 and Chipsec 1.3.6).  Please reopen issue if you are still experiencing difficulties.

closed 2019-01-05 by BrentHoltsclaw