chipsec/chipsec #169: [Errno 2] No such file or directory: '/dev/chipsec'

SaiGonSeamus, 2017-03-12:
Hi,

I'm seeing this on MacOS Sierra. 
I am running as root.
Any suggestions?

Cheers
S

```
[CHIPSEC] Version 1.2.5
[CHIPSEC] Arguments: 
ERROR: Could not start the OS Helper, are you running as Admin/root?
           Message: "Unable to open the Chipsec device.
[Errno 2] No such file or directory: '/dev/chipsec'"
```

c7zero, 2017-03-13:
It means chipsec kernel module (chipsec.kext) wasn't loaded.

After installing chipsec via "pip install chipsec", you'll need to load the kernel module.
Please see here: https://github.com/chipsec/chipsec/tree/master/drivers/osx

You'll need to turn the System Integrity Protection off. This page describes how: https://developer.apple.com/library/mac/documentation/Security/Conceptual/System_Integrity_Protection_Guide/ConfiguringSystemIntegrityProtection/ConfiguringSystemIntegrityProtection.html
Then reboot and load the chipsec kernel module:
   # kextutil chipsec.kext

Mac OS support in chipsec is in alpha, not everything is implemented. Alternatively, you can boot into USB thumb drive with Linux and chipsec installed (via PIP install) and test EFI firmware from there.

c7zero, 2017-03-14:
This should be resolved by PR #175 

Steps:
1. Clone/download chipsec trunk
2. Disable SIP per above instructions
3. From chipsec-master path, run "python setup.py build_ext -i"
    (should complete without errors and should see file chipsec/helper/osx/chipsec.kext)
4. Then can use chipsec, e.g. "sudo python chipsec_util.py spi dump spi.bin"

You need to have XCode installed on Sierra.

closed 2017-03-16 by c7zero