chipsec/chipsec #170: macOS-related weird errors

pwnsdx, 2017-03-12:
Hello,

I tried chipsec on my iMac from 2011 and 2015 (SIP was completely disabled then I tried again with "csrutil --without kext" with no success) and on both of these I got the following:

iMac Mid 2011:

```
[CHIPSEC] ***************************  SUMMARY  ***************************
[CHIPSEC] Time elapsed          0.743
[CHIPSEC] Modules total         17
[CHIPSEC] Modules failed to run 7:
ERROR: chipsec.modules.common.bios_smi
ERROR: chipsec.modules.common.ia32cfg
ERROR: chipsec.modules.common.smrr
ERROR: chipsec.modules.common.secureboot.variables
ERROR: chipsec.modules.common.uefi.access_uefispec
ERROR: chipsec.modules.common.uefi.s3bootscript
ERROR: chipsec.modules.smm_dma
[CHIPSEC] Modules passed        7:
[+] PASSED: chipsec.modules.common.bios_kbrd_buffer
[+] PASSED: chipsec.modules.common.bios_ts
[+] PASSED: chipsec.modules.common.smm
[+] PASSED: chipsec.modules.common.spi_fdopss
[+] PASSED: chipsec.modules.common.spi_lock
[+] PASSED: chipsec.modules.memconfig
[+] PASSED: chipsec.modules.remap
[CHIPSEC] Modules failed        2:
[-] FAILED: chipsec.modules.common.bios_wp
[-] FAILED: chipsec.modules.common.spi_desc
[CHIPSEC] Modules with warnings 1:
[!] WARNING: chipsec.modules.common.rtclock
[CHIPSEC] Modules skipped 0:
[CHIPSEC] Modules with Exceptions 7:
ERROR: chipsec.modules.common.bios_smi
ERROR: chipsec.modules.common.ia32cfg
ERROR: chipsec.modules.common.smrr
ERROR: chipsec.modules.common.secureboot.variables
ERROR: chipsec.modules.common.uefi.access_uefispec
ERROR: chipsec.modules.common.uefi.s3bootscript
ERROR: chipsec.modules.smm_dma
[CHIPSEC] *****************************************************************
```

iMac Late 2015:

```
[CHIPSEC] ***************************  SUMMARY  ***************************
[CHIPSEC] Time elapsed          0.007
[CHIPSEC] Modules total         17
[CHIPSEC] Modules failed to run 7:
ERROR: chipsec.modules.common.bios_smi
ERROR: chipsec.modules.common.ia32cfg
ERROR: chipsec.modules.common.smrr
ERROR: chipsec.modules.common.secureboot.variables
ERROR: chipsec.modules.common.uefi.access_uefispec
ERROR: chipsec.modules.common.uefi.s3bootscript
ERROR: chipsec.modules.smm_dma
[CHIPSEC] Modules passed        5:
[+] PASSED: chipsec.modules.common.bios_kbrd_buffer
[+] PASSED: chipsec.modules.common.bios_ts
[+] PASSED: chipsec.modules.common.smm
[+] PASSED: chipsec.modules.common.spi_fdopss
[+] PASSED: chipsec.modules.common.spi_lock
[CHIPSEC] Modules failed        2:
[-] FAILED: chipsec.modules.common.bios_wp
[-] FAILED: chipsec.modules.common.spi_desc
[CHIPSEC] Modules with warnings 0:
[CHIPSEC] Modules skipped 3:
[*] SKIPPED: chipsec.modules.common.rtclock
[*] SKIPPED: chipsec.modules.memconfig
[*] SKIPPED: chipsec.modules.remap
[CHIPSEC] Modules with Exceptions 7:
ERROR: chipsec.modules.common.bios_smi
ERROR: chipsec.modules.common.ia32cfg
ERROR: chipsec.modules.common.smrr
ERROR: chipsec.modules.common.secureboot.variables
ERROR: chipsec.modules.common.uefi.access_uefispec
ERROR: chipsec.modules.common.uefi.s3bootscript
ERROR: chipsec.modules.smm_dma
[CHIPSEC] *****************************************************************
```

So chipsec.modules.common.bios_wp and chipsec.modules.common.spi_desc failed, here is the details:

```
[*] running module: chipsec.modules.common.bios_wp
[x][ =======================================================================
[x][ Module: BIOS Region Write Protection
[x][ =======================================================================
[*] BC = 0x00 << BIOS Control (b:d.f 00:31.0 + 0xDC)
    [00] BIOSWE           = 0 << BIOS Write Enable 
    [01] BLE              = 0 << BIOS Lock Enable 
    [02] SRC              = 0 << SPI Read Configuration 
    [04] TSS              = 0 << Top Swap Status 
    [05] SMM_BWP          = 0 << SMM BIOS Write Protection 
[-] BIOS region write protection is disabled!
```

```
[*] running module: chipsec.modules.common.spi_desc
[*] Module path: /usr/local/lib/python2.7/site-packages/chipsec/modules/common/spi_desc.pyc
[x][ =======================================================================
[x][ Module: SPI Flash Region Access Control
[x][ =======================================================================
[*] FRAP = 0x0000FFFF << SPI Flash Regions Access Permissions Register (SPIBAR + 0x50)
    [00] BRRA             = FF << BIOS Region Read Access 
    [08] BRWA             = FF << BIOS Region Write Access 
    [16] BMRAG            = 0 << BIOS Master Read Access Grant 
    [24] BMWAG            = 0 << BIOS Master Write Access Grant 
[*] Software access to SPI flash regions: read = 0xFF, write = 0xFF
[-] Software has write access to SPI flash descriptor

[-] FAILED: SPI flash permissions allow SW to write flash descriptor
```

So... is it just a false positive or is there a much bigger trouble there?

c7zero, 2017-03-13:
I'm not sure why some modules returned ERROR without the full log (preferably with --verbose option). For the FAILED modules, I'd suggest to update the firmware to the latest revision and re-run

pwnsdx, 2017-03-14:
Okay so I re-run with verbose, here is the full logs:

iMac 2015:

```
[*] Ignoring unsupported platform warning and continue execution
################################################################
##                                                            ##
##  CHIPSEC: Platform Hardware Security Assessment Framework  ##
##                                                            ##
################################################################
[CHIPSEC] Version 1.2.5
[CHIPSEC] Arguments: -i --verbose
[helper] OSX Helper created
Module chipsec.kext loaded successfully
[helper] OSX Helper started/loaded
[CHIPSEC] API mode: using CHIPSEC kernel module API
[pci] reading B/D/F: 0/0/0, offset: 0x00, value: 0x16108086
[*] looking for platform config in '/Users/sabri/Desktop/chipsec-master/chipsec/cfg/common.xml'..
[*] loading common platform config from '/Users/sabri/Desktop/chipsec-master/chipsec/cfg/common.xml'..
[*] loading integrated devices/controllers..
    + HOSTCTRL        : {'fun': '0', 'bus': '0', 'dev': '0x00', 'vid': '0x8086'}
    + PEG10           : {'fun': '0', 'bus': '0', 'dev': '0x01', 'vid': '0x8086'}
    + PEG11           : {'fun': '1', 'bus': '0', 'dev': '0x01', 'vid': '0x8086'}
    + PEG12           : {'fun': '2', 'bus': '0', 'dev': '0x01', 'vid': '0x8086'}
    + IGD             : {'fun': '0', 'bus': '0', 'dev': '0x02', 'vid': '0x8086'}
    + HDAUDIO         : {'fun': '0', 'bus': '0', 'dev': '0x03', 'vid': '0x8086'}
    + XHCI            : {'fun': '0', 'bus': '0', 'dev': '0x14', 'vid': '0x8086'}
    + MEI1            : {'fun': '0', 'bus': '0', 'dev': '0x16', 'vid': '0x8086'}
    + MEI2            : {'fun': '1', 'bus': '0', 'dev': '0x16', 'vid': '0x8086'}
    + IDER            : {'fun': '2', 'bus': '0', 'dev': '0x16', 'vid': '0x8086'}
    + KT              : {'fun': '3', 'bus': '0', 'dev': '0x16', 'vid': '0x8086'}
    + GBE             : {'fun': '0', 'bus': '0', 'dev': '0x19', 'vid': '0x8086'}
    + EHCI2           : {'fun': '0', 'bus': '0', 'dev': '0x1A', 'vid': '0x8086'}
    + AUDIO           : {'fun': '0', 'bus': '0', 'dev': '0x1B', 'vid': '0x8086'}
    + PCIE1           : {'fun': '0', 'bus': '0', 'dev': '0x1C', 'vid': '0x8086'}
    + PCIE2           : {'fun': '1', 'bus': '0', 'dev': '0x1C', 'vid': '0x8086'}
    + PCIE3           : {'fun': '2', 'bus': '0', 'dev': '0x1C', 'vid': '0x8086'}
    + PCIE4           : {'fun': '3', 'bus': '0', 'dev': '0x1C', 'vid': '0x8086'}
    + PCIE5           : {'fun': '4', 'bus': '0', 'dev': '0x1C', 'vid': '0x8086'}
    + PCIE6           : {'fun': '5', 'bus': '0', 'dev': '0x1C', 'vid': '0x8086'}
    + PCIE7           : {'fun': '6', 'bus': '0', 'dev': '0x1C', 'vid': '0x8086'}
    + PCIE8           : {'fun': '7', 'bus': '0', 'dev': '0x1C', 'vid': '0x8086'}
    + EHCI1           : {'fun': '0', 'bus': '0', 'dev': '0x1D', 'vid': '0x8086'}
    + LPC             : {'fun': '0', 'bus': '0', 'dev': '0x1F', 'vid': '0x8086'}
    + SATA1           : {'fun': '2', 'bus': '0', 'dev': '0x1F', 'vid': '0x8086'}
    + SMBUS           : {'fun': '3', 'bus': '0', 'dev': '0x1F', 'vid': '0x8086'}
    + SATA2           : {'fun': '5', 'bus': '0', 'dev': '0x1F', 'vid': '0x8086'}
    + THERMAL         : {'fun': '6', 'bus': '0', 'dev': '0x1F', 'vid': '0x8086'}
[*] loading MMIO BARs..
    + PXPEPBAR        : {'enable_bit': '0', 'bus': '0', 'mask': '0x7FFFFFF000', 'dev': '0', 'width': '8', 'fun': '0', 'size': '0x1000', 'reg': '0x40', 'desc': 'PCI Express Egress Port Register Range'}
    + MCHBAR          : {'enable_bit': '0', 'bus': '0', 'mask': '0x7FFFFF8000', 'dev': '0', 'width': '8', 'fun': '0', 'size': '0x8000', 'reg': '0x48', 'desc': 'Host Memory Mapped Register Range'}
    + MMCFG           : {'enable_bit': '0', 'bus': '0', 'mask': '0x7FFFFFF000', 'dev': '0', 'width': '8', 'fun': '0', 'size': '0x1000', 'reg': '0x60', 'desc': 'PCI Express Register Range'}
    + DMIBAR          : {'enable_bit': '0', 'bus': '0', 'mask': '0x7FFFFFF000', 'dev': '0', 'width': '8', 'fun': '0', 'size': '0x1000', 'reg': '0x68', 'desc': 'Root Complex Register Range'}
    + GTTMMADR        : {'bus': '0', 'mask': '0x7FFFC00000', 'dev': '2', 'width': '8', 'fun': '0', 'reg': '0x10', 'desc': 'Graphics Translation Table Range'}
    + GMADR           : {'bus': '0', 'mask': '0x7FF8000000', 'dev': '2', 'width': '8', 'fun': '0', 'reg': '0x18', 'desc': 'Graphics Memory Range'}
    + HDABAR          : {'bus': '0', 'mask': '0x7FFFFFF000', 'dev': '3', 'width': '8', 'fun': '0', 'size': '0x1000', 'reg': '0x10', 'desc': 'HD Audio Controller Register Range'}
    + HDBAR           : {'bus': '0', 'mask': '0x7FFFFFC000', 'dev': '0x1B', 'width': '8', 'fun': '0', 'size': '0x1000', 'reg': '0x10', 'desc': 'PCH HD Audio Controller Register Range'}
    + RCBA            : {'enable_bit': '0', 'bus': '0', 'mask': '0xFFFFC000', 'dev': '0x1F', 'width': '4', 'fun': '0', 'size': '0x4000', 'reg': '0xF0', 'desc': 'PCH Root Complex Register Range'}
    + SPIBAR          : {'enable_bit': '0', 'bus': '0', 'mask': '0xFFFFC000', 'dev': '0x1F', 'width': '4', 'offset': '0x3800', 'fun': '0', 'size': '0x200', 'reg': '0xF0', 'desc': 'SPI Controller Register Range'}
    + RCBA_RTC        : {'enable_bit': '0', 'bus': '0', 'mask': '0xFFFFC000', 'dev': '0x1F', 'width': '4', 'offset': '0x3400', 'fun': '0', 'size': '0x200', 'reg': '0xF0', 'desc': 'General Control Register Range'}
    + VTBAR           : {'base_field': 'Base', 'enable_field': 'Enable', 'desc': 'Intel VT-d Register Register Range', 'register': 'VTBAR', 'size': '0x1000'}
[... cut ...]
[*] SKIPPED: chipsec.modules.remap
[CHIPSEC] Modules with Exceptions 7:
ERROR: chipsec.modules.common.bios_smi
ERROR: chipsec.modules.common.ia32cfg
ERROR: chipsec.modules.common.smrr
ERROR: chipsec.modules.common.secureboot.variables
ERROR: chipsec.modules.common.uefi.access_uefispec
ERROR: chipsec.modules.common.uefi.s3bootscript
ERROR: chipsec.modules.smm_dma
[CHIPSEC] *****************************************************************
[helper] OSX Helper stopped/unloaded
[helper] OSX Helper deleted
```

c7zero, 2017-03-14:
ERROR results in modules are due to certain functions not implemented in macOS helper yet (cpuid and EFI_supported). Will be adding these functions... For now, I'd recommend booting off of USB drive with Linux and chipsec.

c7zero, 2017-03-19:
Due to PR #184 modules which raised exceptions will be now skipped when run on macOS. I'll open a new issue to implement missing functionality these modules rely on so that they can run on macOS.

c7zero, 2017-03-19:
Issue #149 tracks this

pwnsdx, 2017-03-19:
Thanks you but what about the FAILED then? I'm unsure using Linux will fix this but I could try when I'll have the time.

c7zero, 2017-03-19:
FAILED modules will also be FAILED when you run under Linux. The two modules returned FAILED because they didn't find specific protections they were looking for. common.bios_wp module failed because your Mac doesn't use SMM based protection of EFI firmware. Instead, it apears to be using PR1-PR2 protected ranges (see the second part of bios_wp log) but they don't cover entire EFI region. common.spi_desc module failed because the system doesn't seem to be restricting access to flash descriptor region via permissions defined in the descriptor itself. Based on the log, your Mac appears to be "soft" protection (PR0 range) to protect flash descriptor. This may be ok but without further analysis I cannot tell if it's a sufficient protection or the firmware can be compromised/corrupted.

pwnsdx, 2017-03-20:
Thanks you but considering that is it probably the same results on every Mac I think your script might need some adjustments.

c7zero, 2017-03-21:
I have a Mac with different results. 
Each module checks for individual protection (or a vulnerability) present/missing on a system and fails if cannot find it. However, the system may be using some alternative protection mechanism which is either checked by some other module, in which case other module will pass, or CHIPSEC doesn't known about this protection, in which case you'd need to do some manual checking. For example, a system may be using write-protection of a SPI device on a device itself instead of using chipset based mechanisms. The sure way to test for all these is attempt to write to SPI chip and see what happens. You can easily do that with CHIPSEC (chipsec_util spi write) but you are risking to brick your system so we cannot have an automatically running module doing this.
Thanks for suggestion! It makes sense to add notes whenever possible about alternative ways to protect the firmware. Closing the issue.

closed 2017-03-21 by c7zero