chipsec/chipsec #174: Doesn't work on Macs from EFI shell

brokeasjoke, 2017-03-14:
Format USB drive as FAT

Download refit EFI shell
http://downloads.sourceforge.net/refit/rEFIt-0.14.dmg?use_mirror=

Copy refit/efi/tools/shell.efi to USB/efi/bootx64.efi

Setup chipsec folders per instructions in manual

Boot Mac holding option

Select "EFI Boot" drive

fs3:\chipsec-master-broke> python chipsec_util.py spi read 0 0x10
################################################################                
##                                                            ##                
##  CHIPSEC: Platform Hardware Security Assessment Framework  ##                
##                                                            ##                
################################################################                
[CHIPSEC] Version 1.2.5                                                         
[CHIPSEC] API mode: using CHIPSEC kernel module API                             
[CHIPSEC] Executing command 'spi' with args ['read', '0', '0x10']               
                                                                                
[CHIPSEC] reading 0x10 bytes from SPI Flash starting at FLA = 0x0               
[CHIPSEC] it may take a few minutes (use DEBUG or VERBOSE logger options to see 
[spi] reading 0x10 bytes from SPI at FLA = 0x0 (in 4 0x4-byte chunks + 0x0-byte 
ERROR: [efi] unsupported size 1 by write_mmio_reg
ERROR: [efi] unsupported size 1 by write_mmio_reg
ERROR: [efi] unsupported size 1 by write_mmio_reg
ERROR: [efi] unsupported size 1 by write_mmio_reg
ERROR: [efi] unsupported size 1 by write_mmio_reg
ERROR: [efi] unsupported size 1 by write_mmio_reg
ERROR: [efi] unsupported size 1 by write_mmio_reg
ERROR: [efi] unsupported size 1 by write_mmio_reg
ERROR: [efi] unsupported size 1 by write_mmio_reg
ERROR: [efi] unsupported size 1 by write_mmio_reg
ERROR: [efi] unsupported size 1 by write_mmio_reg
ERROR: [efi] unsupported size 1 by write_mmio_reg
ERROR: [efi] unsupported size 1 by write_mmio_reg
7c ff ff ff 7c ff ff ff 7c ff ff ff 7c ff ff ff | |   |   |   |   
[CHIPSEC] completed SPI flash memory read
[CHIPSEC] (spi read) time elapsed 0.000

brokeasjoke, 2017-03-14:
Also tried with release v 1.2.5 same results

c7zero, 2017-03-14:
Can you do the same but with --verbose option and reading 4kB ?
python chipsec_util.py --verbose spi read 0 0x1000

brokeasjoke, 2017-03-14:
attached

[log.txt](https://github.com/chipsec/chipsec/files/842343/log.txt)

brokeasjoke, 2017-03-14:
Was tested from a MacBookAir7,2

dukechem, 2017-03-15:
I had same errors, but the chipsec-shellfix got rid of the errors. (From refit shell.efi on my Mac). See:
[https://github.com/chipsec/chipsec/pull/178#issuecomment-286857772](https://github.com/chipsec/chipsec/pull/178#issuecomment-286857772)