chipsec/chipsec #183: chipsec freezes from uefi shell

hydrajump, 2017-03-16:
Booted from USB stick with uefi shell and chipsec on Mac.

Boots fine and then I select the USB stick and try to run chipsec:

```
fs3:
cd EFI\Chipsec
python chipsec_main.py
...
[x][ Module: Pre-boot Passwords in the BIOS Keyboard Buffer
[x][ =====================================
```
It just freezes at this point.

Similar freeze happens if I try:

```
fs3:
cd EFI\Chipsec
python chipsec_util.py acpi list
...
[CHIPSEC] Executing command 'acpi' with args [list]
...

Why is it freezing?

c7zero, 2017-03-16:
Does --verbose say anything more before the hang?
python chipsec_main.py --verbose

It may be incompatible UEFI shell version with Mac EFI firmware...

hydrajump, 2017-03-16:
@c7zero just ran `python chipsec_main.py --verbose`

```
...
[x][ Module: Pre-boot Passwords in the BIOS Keyboard Buffer
[x][ =====================================
[mem] 0x00000000000041A
[efi] helper does not support 64b PA
```

hydrajump, 2017-03-17:
I tried using the same refit 0.14 shell.efi as mentioned in [this] issue (https://github.com/chipsec/chipsec/issues/174) and I get the same freeze and error message :(

andrew-furtak, 2017-03-22:
What model is your mac? I'm trying to run cihpsec on Macbook Air mid 2013 and don't see the issue. I'm using the shell from the 1st link (https://github.com/tianocore/edk2/blob/master/EdkShellBinPkg/FullShell/X64/Shell_Full.efi).

\>ver
EFI Specification Revision : 1.10
EFI Vendor                 : Apple
EFI Revision               : 1.10

hydrajump, 2017-03-22:
@andrew-furtak the error above was when testing on a MacBook Pro 2016 model.

closed 2021-11-23 by npmitche