chipsec/chipsec #186: chipsec support for legacy BIOS and UEFI-compatibility mode boot.

ismaws, 2017-03-20:
Currently, chipsec efi symbol is missing on OS with BIOS and UEFI-compatibility mode boot. We need to workaround the efi structure to compile the driver successfully.

chipsec driver build errors:

WARNING: "efi" [/home/user/git/chipsec/build/lib.linux-x86_64-2.7/drivers/linux/chipsec.ko] undefined!

abazhaniuk, 2017-04-20:
what version of chipsec did you test? 
what version OS and kernel you have? 
i've tested latest chipsec on kali with legacy BIOS and chipsec works fine.

DropBearWare, 2018-05-16:
I think I'm hitting the same deal.

-- build log --
`root@ubuntu:/home/sds/chipsec# python setup.py build_ext -i
running build_ext
copying build/lib.linux-x86_64-2.7/chipsec/helper/linux/cores.so -> chipsec/helper/linux
copying build/lib.linux-x86_64-2.7/chipsec_tools/efi_compressor.so -> chipsec_tools
building the linux driver
creating /home/sds/chipsec/build/lib.linux-x86_64-2.7/drivers
creating /home/sds/chipsec/build/lib.linux-x86_64-2.7/drivers/linux
copying drivers/linux/COPYING -> /home/sds/chipsec/build/lib.linux-x86_64-2.7/drivers/linux
copying drivers/linux/Makefile -> /home/sds/chipsec/build/lib.linux-x86_64-2.7/drivers/linux
copying drivers/linux/dkms.conf -> /home/sds/chipsec/build/lib.linux-x86_64-2.7/drivers/linux
copying drivers/linux/AUTHORS -> /home/sds/chipsec/build/lib.linux-x86_64-2.7/drivers/linux
creating /home/sds/chipsec/build/lib.linux-x86_64-2.7/drivers/linux/include
copying drivers/linux/include/chipsec.h -> /home/sds/chipsec/build/lib.linux-x86_64-2.7/drivers/linux/include
copying drivers/linux/WARNING.txt -> /home/sds/chipsec/build/lib.linux-x86_64-2.7/drivers/linux
copying drivers/linux/README -> /home/sds/chipsec/build/lib.linux-x86_64-2.7/drivers/linux
creating /home/sds/chipsec/build/lib.linux-x86_64-2.7/drivers/linux/i386
copying drivers/linux/i386/cpu.asm -> /home/sds/chipsec/build/lib.linux-x86_64-2.7/drivers/linux/i386
copying drivers/linux/chipsec_km.c -> /home/sds/chipsec/build/lib.linux-x86_64-2.7/drivers/linux
creating /home/sds/chipsec/build/lib.linux-x86_64-2.7/drivers/linux/amd64
copying drivers/linux/amd64/cpu.asm -> /home/sds/chipsec/build/lib.linux-x86_64-2.7/drivers/linux/amd64
WARNING: could not find /home/sds/chipsec/build/lib.linux-x86_64-2.7/drivers/linux/amd64/.cpu.o.cmd for /home/sds/chipsec/build/lib.linux-x86_64-2.7/drivers/linux/amd64/cpu.o
WARNING: "efi" [/home/sds/chipsec/build/lib.linux-x86_64-2.7/drivers/linux/chipsec.ko] undefined!
copying /home/sds/chipsec/build/lib.linux-x86_64-2.7/drivers/linux/chipsec.ko -> chipsec/helper/linux
removing '/home/sds/chipsec/build/lib.linux-x86_64-2.7/drivers' (and everything under it)
`
-- execution --
root@ubuntu:/home/sds/chipsec# python chipsec_main.py
################################################################
##                                                            ##
##  CHIPSEC: Platform Hardware Security Assessment Framework  ##
##                                                            ##
################################################################
[CHIPSEC] Version 1.3.6rc1
[CHIPSEC] Arguments:
insmod: ERROR: could not insert module /home/sds/chipsec/chipsec/helper/linux/chipsec.ko: Unknown symbol in module
ERROR: Could not start the OS Helper, are you running as Admin/root?
           Message: "Command '['insmod', '/home/sds/chipsec/chipsec/helper/linux/chipsec.ko', '', '']' returned non-zero exit status 1"

-- system details --
root@ubuntu:/home/sds/chipsec# lsb_release -a
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 16.04.4 LTS
Release:        16.04
Codename:       xenial

DropBearWare, 2018-05-16:
on request from @BrentHoltsclaw here's more detailing..

`
root@ubuntu:/home/sds/chipsec# find / -type f -name "chipsec.ko"
/home/sds/chipsec/chipsec/helper/linux/chipsec.ko
/home/sds/chipsec/build/lib.linux-x86_64-2.7/chipsec/helper/linux/chipsec.ko
/root/anaconda2/lib/python2.7/site-packages/chipsec-1.3.6rc1-py2.7-linux-x86_64.egg/chipsec/helper/linux/chipsec.ko
root@ubuntu:/home/sds/chipsec# insmod /home/sds/chipsec/chipsec/helper/linux/chipsec.ko
insmod: ERROR: could not insert module /home/sds/chipsec/chipsec/helper/linux/chipsec.ko: Unknown symbol in module
root@ubuntu:/home/sds/chipsec# insmod /root/anaconda2/lib/python2.7/site-packages/chipsec-1.3.6rc1-py2.7-linux-x86_64.egg/chipsec/helper/linux/chipsec.ko
insmod: ERROR: could not insert module /root/anaconda2/lib/python2.7/site-packages/chipsec-1.3.6rc1-py2.7-linux-x86_64.egg/chipsec/helper/linux/chipsec.ko: Unknown symbol in module
root@ubuntu:/home/sds/chipsec# insmod /home/sds/chipsec/build/lib.linux-x86_64-2.7/chipsec/helper/linux/chipsec.ko
insmod: ERROR: could not insert module /home/sds/chipsec/build/lib.linux-x86_64-2.7/chipsec/helper/linux/chipsec.ko: Unknown symbol in module

root@ubuntu:/home/sds/chipsec# dmesg | tail
[  255.978357] chipsec: loading out-of-tree module taints kernel.
[  255.978424] chipsec: module verification failed: signature and/or required key missing - tainting kernel
[  255.978506] chipsec: Unknown symbol efi (err 0)
[  287.368104] chipsec: Unknown symbol efi (err 0)
[  502.495551] chipsec: Unknown symbol efi (err 0)
`

firewareie, 2019-04-20:
I got a problem:
insmod: ERROR: could not insert module /usr/local/lib/python2.7/dist-packages/chipsec-1.3.7-py2.7-linux-x86_64.egg/chipsec/helper/linux/chipsec.ko: Required key not available
-- system details --
Linux iie-HP-Zhan-86-Pro-G2-MT-Business-PC 4.15.0-45-generic #48~16.04.1-Ubuntu SMP Tue Jan 29 18:03:48 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux

BrentHoltsclaw, 2019-04-22:
@firewareie It appears that you are using secure boot.  There are a few options, you can disable secure boot, disable driver validation with mokutil, or sign the CHIPSEC module and register the keys using mokutil.  See [instructions](https://askubuntu.com/questions/762254/why-do-i-get-required-key-not-available-when-install-3rd-party-kernel-modules) here.