chipsec/chipsec #193: Failed: chipsec.modules.common.eufi.access.uefispec

Sir-Stier, 2017-04-03:
I'm receiving failures with a few error responses:

Failed: Some EFI variables were not protected according to spec.

Variables that differ from UEFI spec:
Conout
Conout
SignatureSupport
SetupMode
Conin
Conin

Didn't find any S3 boot-scripts in EFI variables
WARNING S3 Boot-Script was not found. Firmware May be using other ways to store....

[CHIPSEC] Modules failed  1:
FAILED chipsec.modules.common.uefi.access.uefispec

WARNING chipset.modules.common.rtclock
WARNING chipset.modules.common.secureboot.variables
WARNING chipset.modules.common.uefi.s3bootscript

For reference I'm using an HP Elitebook. Without secureboot enabled. 

I've looking into enabling secureboot and moving from setup mode to usermode. Not sure why S3 boot wouldn't be found.

ErikBjorge, 2017-04-03:
Warnings just indicate further investigation is required.  In the case of the S3 boot scripts, chipsec is not able to find them and is thus not able to determine if they are correctly protected from modification.  They may be protected but the tool just can't determine if they are.

Can you run chipsec_main with the -v option and attach the log so we can investigate the warnings further?

Sir-Stier, 2017-04-12:
I ran Chipset on  similar laptop purchased in the same batch. It's unable to find the S3 BootScript after the most recent BIOS update.

I'll be able to resolve one of the failures through enabling SecureBoot (which required reformatting.)

Wondering if it's possible to dig into the variables and resolve the failure mention?

chipsec.modules.common.uefi.access_uefispec

Thanks in advance!

################################################################
##                                                            ##
##  CHIPSEC: Platform Hardware Security Assessment Framework  ##
##                                                            ##
################################################################
[CHIPSEC] Version 1.3.0
[CHIPSEC] Arguments: -v
[helper] UEFI Helper created
[helper] UEFI Helper started/loaded
[CHIPSEC] API mode: using CHIPSEC kernel module API
[pci] reading B/D/F: 0/0/0, offset: 0x00, value: 0x01548086
[*] Couldn't import chipsec.cfg.ivb
No module named ivb
[*] looking for platform config in 'fs1:\chipsec\chipsec/cfg\common.xml'..
[*] loading common platform config from 'fs1:\chipsec\chipsec/cfg\common.xml'..
[*] loading integrated devices/controllers..
    + HOSTCTRL        : {'fun': '0', 'bus': '0', 'dev': '0x00', 'vid': '0x8086'}
    + PEG10           : {'fun': '0', 'bus': '0', 'dev': '0x01', 'vid': '0x8086'}
    + PEG11           : {'fun': '1', 'bus': '0', 'dev': '0x01', 'vid': '0x8086'}
    + PEG12           : {'fun': '2', 'bus': '0', 'dev': '0x01', 'vid': '0x8086'}
    + IGD             : {'fun': '0', 'bus': '0', 'dev': '0x02', 'vid': '0x8086'}
    + HDAUDIO         : {'fun': '0', 'bus': '0', 'dev': '0x03', 'vid': '0x8086'}
    + XHCI            : {'fun': '0', 'bus': '0', 'dev': '0x14', 'vid': '0x8086'}
    + MEI1            : {'fun': '0', 'bus': '0', 'dev': '0x16', 'vid': '0x8086'}
    + MEI2            : {'fun': '1', 'bus': '0', 'dev': '0x16', 'vid': '0x8086'}
    + IDER            : {'fun': '2', 'bus': '0', 'dev': '0x16', 'vid': '0x8086'}
    + KT              : {'fun': '3', 'bus': '0', 'dev': '0x16', 'vid': '0x8086'}
    + GBE             : {'fun': '0', 'bus': '0', 'dev': '0x19', 'vid': '0x8086'}
    + EHCI2           : {'fun': '0', 'bus': '0', 'dev': '0x1A', 'vid': '0x8086'}
    + AUDIO           : {'fun': '0', 'bus': '0', 'dev': '0x1B', 'vid': '0x8086'}
    + PCIE1           : {'fun': '0', 'bus': '0', 'dev': '0x1C', 'vid': '0x8086'}
    + PCIE2           : {'fun': '1', 'bus': '0', 'dev': '0x1C', 'vid': '0x8086'}
    + PCIE3           : {'fun': '2', 'bus': '0', 'dev': '0x1C', 'vid': '0x8086'}
    + PCIE4           : {'fun': '3', 'bus': '0', 'dev': '0x1C', 'vid': '0x8086'}
    + PCIE5           : {'fun': '4', 'bus': '0', 'dev': '0x1C', 'vid': '0x8086'}
    + PCIE6           : {'fun': '5', 'bus': '0', 'dev': '0x1C', 'vid': '0x8086'}
    + PCIE7           : {'fun': '6', 'bus': '0', 'dev': '0x1C', 'vid': '0x8086'}
    + PCIE8           : {'fun': '7', 'bus': '0', 'dev': '0x1C', 'vid': '0x8086'}
    + EHCI1           : {'fun': '0', 'bus': '0', 'dev': '0x1D', 'vid': '0x8086'}
    + LPC             : {'fun': '0', 'bus': '0', 'dev': '0x1F', 'vid': '0x8086'}
    + SATA1           : {'fun': '2', 'bus': '0', 'dev': '0x1F', 'vid': '0x8086'}
    + SMBUS           : {'fun': '3', 'bus': '0', 'dev': '0x1F', 'vid': '0x8086'}
    + SATA2           : {'fun': '5', 'bus': '0', 'dev': '0x1F', 'vid': '0x8086'}
    + THERMAL         : {'fun': '6', 'bus': '0', 'dev': '0x1F', 'vid': '0x8086'}
[*] loading MMIO BARs..
    + PXPEPBAR        : {'enable_bit': '0', 'bus': '0', 'mask': '0x7FFFFFF000', 'dev': '0', 'width': '8', 'fun': '0', 'size': '0x1000', 'reg': '0x40', 'desc': 'PCI Express Egress Port Register Range'}
    + MCHBAR          : {'enable_bit': '0', 'bus': '0', 'mask': '0x7FFFFF8000', 'dev': '0', 'width': '8', 'fun': '0', 'size': '0x8000', 'reg': '0x48', 'desc': 'Host Memory Mapped Register Range'}
    + MMCFG           : {'enable_bit': '0', 'bus': '0', 'mask': '0x7FFFFFF000', 'dev': '0', 'width': '8', 'fun': '0', 'size': '0x1000', 'reg': '0x60', 'desc': 'PCI Express Register Range'}
    + DMIBAR          : {'enable_bit': '0', 'bus': '0', 'mask': '0x7FFFFFF000', 'dev': '0', 'width': '8', 'fun': '0', 'size': '0x1000', 'reg': '0x68', 'desc': 'Root Complex Register Range'}
    + GTTMMADR        : {'bus': '0', 'mask': '0x7FFFC00000', 'dev': '2', 'width': '8', 'fun': '0', 'reg': '0x10', 'desc': 'Graphics Translation Table Range'}
    + GMADR           : {'bus': '0', 'mask': '0x7FF8000000', 'dev': '2', 'width': '8', 'fun': '0', 'reg': '0x18', 'desc': 'Graphics Memory Range'}
    + HDABAR          : {'bus': '0', 'mask': '0x7FFFFFF000', 'dev': '3', 'width': '8', 'fun': '0', 'size': '0x1000', 'reg': '0x10', 'desc': 'HD Audio Controller Register Range'}
    + HDBAR           : {'bus': '0', 'mask': '0x7FFFFFC000', 'dev': '0x1B', 'width': '8', 'fun': '0', 'size': '0x1000', 'reg': '0x10', 'desc': 'PCH HD Audio Controller Register Range'}
    + RCBA            : {'enable_bit': '0', 'bus': '0', 'mask': '0xFFFFC000', 'dev': '0x1F', 'width': '4', 'fun': '0', 'size': '0x4000', 'reg': '0xF0', 'desc': 'PCH Root Complex Register Range'}
    + SPIBAR          : {'enable_bit': '0', 'bus': '0', 'mask': '0xFFFFC000', 'dev': '0x1F', 'width': '4', 'offset': '0x3800', 'fun': '0', 'size': '0x200', 'reg': '0xF0', 'desc': 'SPI Controller Register Range'}
[... cut ...]
[+] PASSED: chipsec.modules.remap
[+] PASSED: chipsec.modules.smm_dma
[CHIPSEC] Modules failed        1:
[-] FAILED: chipsec.modules.common.uefi.access_uefispec
[CHIPSEC] Modules with warnings 3:
[!] WARNING: chipsec.modules.common.rtclock
[!] WARNING: chipsec.modules.common.secureboot.variables
[!] WARNING: chipsec.modules.common.uefi.s3bootscript
[CHIPSEC] Modules skipped 0:
[CHIPSEC] *****************************************************************
[helper] UEFI Helper stopped/unloaded
[helper] UEFI Helper deleted

c7zero, 2017-05-09:
[-] FAILED: chipsec.modules.common.uefi.access_uefispec

- You can ignore ConIn/ConOut
- SetupMode seems to be protected by authenticated variables (its updates are supposed to be signed) while Secure Boot in UEFI spec expects it to be modifiable with physical presence only. Not sure at the moment if its broken or ok security wise, need more time to analyze
- SignatureSupport variable is also authenticated on that system. I'm guessing this is a replacement for "read-only" requirement which should have been implemented with VariableLock protocol 

[!] WARNING: chipsec.modules.common.rtclock

This warning is because BIOS isn't locking locations at offsets 0x38-0x3F in both low and high banks of RTC backed memory. In most cases, BIOS isn't using them at all therefore isn't locking them. You can check if there's anything at these offsets by dumping CMOS memory via "chipsec_util cmos dump" command.

[!] WARNING: chipsec.modules.common.secureboot.variables

UEFI Secure Boot is disabled on the system hence the Warning

[!] WARNING: chipsec.modules.common.uefi.s3bootscript

Like @ErikBjorge pointed, chipsec couldn't find S3 Boot Script which likely indicates that your BIOS is storing it in SMM memory inaccessible to software/chipsec so you are likely fine here

c7zero, 2017-06-09:
Closing.. Please re-open if you still have questions left

closed 2017-06-09 by c7zero