chipsec/chipsec #194: Win 10/Chipsec 1.3.0/tools.uefi.uefivar_fuzz - "returned error, Parameter incorrect"

luongthomas, 2017-04-04:
Running on Windows 10.  Other chipsec 1.3.0 modules run fine and return output as expected.

Testing out the new uefivar_fuzz module and the script returns errors for all the tests.

Command:
`chipsec_main.py -m tools.uefi.uefivar_fuzz`

Output:
`Running test # 14:
ERROR: SetFirmwareEnvironmentVariable[Ex] returned error: [Error 87] The parameter is incorrect. 
87
ERROR: SetFirmwareEnvironmentVariable[Ex] returned error: [Error 87] The parameter is incorrect. 
87

Running test # 15:
ERROR: SetFirmwareEnvironmentVariable[Ex] returned error: [Error 87] The parameter is incorrect. 
87
ERROR: SetFirmwareEnvironmentVariable[Ex] returned error: [Error 87] The parameter is incorrect. 
87
`

ErikBjorge, 2017-04-07:
I have been able to reproduce the same failure.  This should be marked as a bug and I will determine the root cause of the failure.

ErikBjorge, 2017-04-08:
I did some investigation into how the uefivar_fuzz module works and the errors are expected.  The fuzzer works by generating random values for the variable name, guid, data, datasize and attributes.  Because the default mode of the tool is to fuzz all of the input parameters at once I would expect either the OS or firmware interface to reject most transactions.  By fuzzing the interface we are looking for cases where the system behaves in an unexpected way (hangs, crashes or fails to boot).

In your specific case the error is being returned by the Windows UEFI variable interface via GetLastError.  I am not sure if the OS or firmware parameter verification code detected the error state.

c7zero, 2017-05-09:
@luongthomas You can try changing any of these to False, especially FUZZ_SIZE and FUZZ_ATTRIB. They are likely the cause of SetVariable API returning parameter incorrect error (basically, rejecting invalid variable).

        FUZZ_NAME   = True 
        FUZZ_GUID   = True
        FUZZ_ATTRIB = True
        FUZZ_DATA   = True
        FUZZ_SIZE   = True

c7zero, 2017-06-09:
Closing.. Please re-open if you still have questions left

closed 2017-06-09 by c7zero