chipsec/chipsec #197: Can't build on 32 bit Kali 4.6.0

ewilded, 2017-04-17:
root@kali:/lib/live/mount/persistence/sdb3# pip install chipsec
Collecting chipsec
  Downloading chipsec-1.3.0.tar.gz (1.0MB)
    100% |████████████████████████████████| 1.0MB 143kB/s 
Building wheels for collected packages: chipsec
  Running setup.py bdist_wheel for chipsec ... error
  Complete output from command /usr/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-lJulPF/chipsec/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" bdist_wheel -d /tmp/tmpZMv9jdpip-wheel- --python-tag cp27:
  running bdist_wheel
  running build
  running build_py
  creating build
  creating build/lib.linux-i686-2.7
  copying chipsec_main.py -> build/lib.linux-i686-2.7
  copying chipsec_util.py -> build/lib.linux-i686-2.7
  creating build/lib.linux-i686-2.7/chipsec_tools
  copying chipsec_tools/__init__.py -> build/lib.linux-i686-2.7/chipsec_tools
  creating build/lib.linux-i686-2.7/chipsec
  copying chipsec/chipset.py -> build/lib.linux-i686-2.7/chipsec
  copying chipsec/command.py -> build/lib.linux-i686-2.7/chipsec
  copying chipsec/module.py -> build/lib.linux-i686-2.7/chipsec
  copying chipsec/logger.py -> build/lib.linux-i686-2.7/chipsec
  copying chipsec/file.py -> build/lib.linux-i686-2.7/chipsec
  copying chipsec/defines.py -> build/lib.linux-i686-2.7/chipsec
  copying chipsec/xmlout.py -> build/lib.linux-i686-2.7/chipsec
  copying chipsec/__init__.py -> build/lib.linux-i686-2.7/chipsec
  copying chipsec/module_common.py -> build/lib.linux-i686-2.7/chipsec
  creating build/lib.linux-i686-2.7/chipsec_tools/linux
  copying chipsec_tools/linux/__init__.py -> build/lib.linux-i686-2.7/chipsec_tools/linux
  creating build/lib.linux-i686-2.7/chipsec_tools/windows
  copying chipsec_tools/windows/__init__.py -> build/lib.linux-i686-2.7/chipsec_tools/windows
  creating build/lib.linux-i686-2.7/chipsec/cfg
  copying chipsec/cfg/common.py -> build/lib.linux-i686-2.7/chipsec/cfg
  copying chipsec/cfg/__init__.py -> build/lib.linux-i686-2.7/chipsec/cfg
  creating build/lib.linux-i686-2.7/chipsec/modules
  copying chipsec/modules/memconfig.py -> build/lib.linux-i686-2.7/chipsec/modules
  copying chipsec/modules/remap.py -> build/lib.linux-i686-2.7/chipsec/modules
  copying chipsec/modules/__init__.py -> build/lib.linux-i686-2.7/chipsec/modules
  copying chipsec/modules/smm_dma.py -> build/lib.linux-i686-2.7/chipsec/modules
  creating build/lib.linux-i686-2.7/chipsec/utilcmd
  copying chipsec/utilcmd/io_cmd.py -> build/lib.linux-i686-2.7/chipsec/utilcmd
  copying chipsec/utilcmd/mem_cmd.py -> build/lib.linux-i686-2.7/chipsec/utilcmd
  copying chipsec/utilcmd/msr_cmd.py -> build/lib.linux-i686-2.7/chipsec/utilcmd
  copying chipsec/utilcmd/mmcfg_cmd.py -> build/lib.linux-i686-2.7/chipsec/utilcmd
  copying chipsec/utilcmd/igd_cmd.py -> build/lib.linux-i686-2.7/chipsec/utilcmd
  copying chipsec/utilcmd/spd_cmd.py -> build/lib.linux-i686-2.7/chipsec/utilcmd
  copying chipsec/utilcmd/chipset_cmd.py -> build/lib.linux-i686-2.7/chipsec/utilcmd
  copying chipsec/utilcmd/msgbus_cmd.py -> build/lib.linux-i686-2.7/chipsec/utilcmd
  copying chipsec/utilcmd/cpu_cmd.py -> build/lib.linux-i686-2.7/chipsec/utilcmd
  copying chipsec/utilcmd/desc_cmd.py -> build/lib.linux-i686-2.7/chipsec/utilcmd
  copying chipsec/utilcmd/acpi_cmd.py -> build/lib.linux-i686-2.7/chipsec/utilcmd
  copying chipsec/utilcmd/uefi_cmd.py -> build/lib.linux-i686-2.7/chipsec/utilcmd
  copying chipsec/utilcmd/pci_cmd.py -> build/lib.linux-i686-2.7/chipsec/utilcmd
  copying chipsec/utilcmd/interrupts_cmd.py -> build/lib.linux-i686-2.7/chipsec/utilcmd
  copying chipsec/utilcmd/ec_cmd.py -> build/lib.linux-i686-2.7/chipsec/utilcmd
  copying chipsec/utilcmd/spi_cmd.py -> build/lib.linux-i686-2.7/chipsec/utilcmd
  copying chipsec/utilcmd/decode_cmd.py -> build/lib.linux-i686-2.7/chipsec/utilcmd
  copying chipsec/utilcmd/ucode_cmd.py -> build/lib.linux-i686-2.7/chipsec/utilcmd
  copying chipsec/utilcmd/cmos_cmd.py -> build/lib.linux-i686-2.7/chipsec/utilcmd
  copying chipsec/utilcmd/smbus_cmd.py -> build/lib.linux-i686-2.7/chipsec/utilcmd
  copying chipsec/utilcmd/mmio_cmd.py -> build/lib.linux-i686-2.7/chipsec/utilcmd
  copying chipsec/utilcmd/__init__.py -> build/lib.linux-i686-2.7/chipsec/utilcmd
  copying chipsec/utilcmd/vmm_cmd.py -> build/lib.linux-i686-2.7/chipsec/utilcmd
  copying chipsec/utilcmd/spidesc_cmd.py -> build/lib.linux-i686-2.7/chipsec/utilcmd
  copying chipsec/utilcmd/iommu_cmd.py -> build/lib.linux-i686-2.7/chipsec/utilcmd
  creating build/lib.linux-i686-2.7/chipsec/helper
  copying chipsec/helper/oshelper.py -> build/lib.linux-i686-2.7/chipsec/helper
  copying chipsec/helper/helpers.py -> build/lib.linux-i686-2.7/chipsec/helper
  copying chipsec/helper/__init__.py -> build/lib.linux-i686-2.7/chipsec/helper
  creating build/lib.linux-i686-2.7/chipsec/hal
  copying chipsec/hal/msgbus.py -> build/lib.linux-i686-2.7/chipsec/hal
[... 628 lines cut ...]
model name	: AMD E-300 APU with Radeon(tm) HD Graphics
stepping	: 0
microcode	: 0x5000101
cpu MHz		: 780.000
cache size	: 512 KB
physical id	: 0
siblings	: 2
core id		: 1
cpu cores	: 2
apicid		: 1
initial apicid	: 1
fdiv_bug	: no
f00f_bug	: no
coma_bug	: no
fpu		: yes
fpu_exception	: yes
cpuid level	: 6
wp		: yes
flags		: fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush mmx fxsr sse sse2 ht syscall nx mmxext fxsr_opt pdpe1gb rdtscp lm constant_tsc rep_good nonstop_tsc extd_apicid aperfmperf eagerfpu pni monitor ssse3 cx16 popcnt lahf_lm cmp_legacy svm extapic cr8_legacy abm sse4a misalignsse 3dnowprefetch ibs skinit wdt hw_pstate vmmcall arat npt lbrv svm_lock nrip_save pausefilter
bugs		: fxsave_leak sysret_ss_attrs
bogomips	: 2594.47
clflush size	: 64
cache_alignment	: 64
address sizes	: 36 bits physical, 48 bits virtual
power management: ts ttp tm stc 100mhzsteps hwpstate

abazhaniuk, 2017-04-19:
reproduced issue. $(ARCH) in Makefile is not defined. and make try to build amd64 asm file on 32 bits system. working on fix.

abazhaniuk, 2017-04-21:
fix in https://github.com/chipsec/chipsec/pull/203 
try to use chipsec from github on you Kali 32bit system.

closed 2017-04-21 by abazhaniuk