chipsec/chipsec #198: Update to versioning of chipsec to clearly identify released versions

ErikBjorge, 2017-04-17:
Problem:
Currently the latest release and the working head revision have the same version string.  This makes determining if a released version is being used difficult.

I would like the version string updated to make the difference between the head version and a release version more clear.

Proposal:
- Release versions keep the original version string (1.3.0).
- Version string is updated to indicate that the head revision is not a released version (1.3.0head)
- Requires that this file is updated before and after each release is generated.

tweksteen, 2017-04-18:
We might be able to do a bit better and automate this step by using a python versioning library: see https://github.com/warner/python-versioneer for instance.

This will also give us more details in case of a bug report as we will know which exact commit (not just head) was used.

ErikBjorge, 2017-04-19:
This looks like a good way of extracting some extra information and including it in the version string.  This will be helpful when debugging as it will point you at the exact version of the code being used.

I think we do need to look at a couple of potential issues.
- This is an additional dependency that needs to be installed on the target system.  Not a deal breaker but something to think about.
- How do we deal with version information when additional modules/tools/configurations are added to chipsec?  We may just want to note that additional content is present.

I think that adding support for python-versioneer into setup.py we are better off even if it does not cover all of the potential use cases.  I will see about adding the support soon.  Thanks for pointing this out to me.

c7zero, 2017-05-09:
Did PR #209 solve this issue? I'm not sure I like adding new dependency either

ErikBjorge, 2018-12-21:
Moving to more frequent releases to mitigate this issue.  Closing, reopen if more discussion is needed.

closed 2018-12-21 by ErikBjorge