chipsec/chipsec #207: Potential UEFI secureboot malware activity?

GreenlightAU, 2017-04-25:
I have a Samsung NP350 laptop, and I recently had a potential rootkit scare after visiting a phishing site by mistake.  I wiped my computer and all HDD partitions to be on the safe side and upon reinstalling Windows 8.1 I found I couldn't install KB3172729 which is an update that modifies the secure boot blacklist.  I'm certain it installed with no issues before the wipe which is why I'm concerned.   I even reflashed the UEFI with Samsung's official UEFI rom but it didn't seem to help.

The blacklist of modules seemed to come up clean, but the other results from chipsec_main seemed to indicate some incompatibilities with the UEFI spec, would anyone have an idea if they are reason to worry?

[-] Variables with attributes that differ from UEFI spec:
    ErrOut
    SignatureSupport
    SetupMode
    SecureBoot

[*] Variable ErrOut (BS+RT)
[!]   Missing attributes:NV
[*] Variable SignatureSupport (BS+RT+AWS)
[!]   Extra attributes:AWS
[*] Variable SecureBoot (BS+RT+AWS)
[!]   Extra attributes:AWS
[!] Found 1 S3 boot-script(s) in EFI variables

[CHIPSEC] Modules failed        3:
[-] FAILED: chipsec.modules.common.bios_smi
[-] FAILED: chipsec.modules.common.bios_wp
[-] FAILED: chipsec.modules.common.uefi.s3bootscript
[CHIPSEC] Modules with warnings 1:
[!] WARNING: chipsec.modules.common.uefi.access_uefispec

c7zero, 2017-05-09:
Well, looking at failed common.bios_wp and common.uefi.s3bootscript modules, it's likely that the UEFI on your system can be compromised. It appears to be vulnerable to one of the S3 boot script issues and doesn't have write-protection enabled. To be sure, you can post full log of these two modules.

Did you run tools.uefi.whitelist module at all to see if there are any extra EFI executables not present in the original firmware image? You can create the original JSON file from original image downloaded from Samsung's web-site. Use "chipsec_main -m tools.uefi.whitelist -a generate,efilist.json,original.rom" to generate the list then "chipsec_main -m tools.uefi.whitelist -a check,efilist.json" to check firmware on the system against the expected.

GreenlightAU, 2017-05-09:
Hi c7zero, thanks very much for responding to my question and for the advice.

I ran the whitelist module as you suggested and it seems to have passed ok with the following messages:
[+] PASSED: all EFI executables match the list
[+] PASSED: chipsec.modules.tools.uefi.whitelist

In case it's of any use, currently the only thing in the secureboot DBX blacklist database is a Samsung entry (most likely from when I reset all the secureboot variables upon trying to fix this issue:
dbx Variable Certificates and Hashes (Forbidden Signers):
X.509 Certificate:	CN=SAMSUNG ELECTRONICS_PK

I'm hoping I'm pasting the correct thing here regarding the other modules:

# common.bios_wp
[*] running module: chipsec.modules.common.bios_wp
[x][ =======================================================================
[x][ Module: BIOS Region Write Protection
[x][ =======================================================================
[*] BC = 0x08 << BIOS Control (b:d.f 00:31.0 + 0xDC)
    [00] BIOSWE           = 0 << BIOS Write Enable 
    [01] BLE              = 0 << BIOS Lock Enable 
    [02] SRC              = 2 << SPI Read Configuration 
    [04] TSS              = 0 << Top Swap Status 
    [05] SMM_BWP          = 0 << SMM BIOS Write Protection 
[-] BIOS region write protection is disabled!

[*] BIOS Region: Base = 0x00180000, Limit = 0x005FFFFF
SPI Protected Ranges
------------------------------------------------------------
PRx (offset) | Value    | Base     | Limit    | WP? | RP?
------------------------------------------------------------
PR0 (74)     | 00000000 | 00000000 | 00000000 | 0   | 0 
PR1 (78)     | 00000000 | 00000000 | 00000000 | 0   | 0 
PR2 (7C)     | 00000000 | 00000000 | 00000000 | 0   | 0 
PR3 (80)     | 00000000 | 00000000 | 00000000 | 0   | 0 
PR4 (84)     | 00000000 | 00000000 | 00000000 | 0   | 0 

[!] None of the SPI protected ranges write-protect BIOS region

[!] BIOS should enable all available SMM based write protection mechanisms or configure SPI protected ranges to protect the entire BIOS region
[-] FAILED: BIOS is NOT protected completely

# common.uefi.s3bootscript
[*] running module: chipsec.modules.common.uefi.s3bootscript
[x][ =======================================================================
[x][ Module: S3 Resume Boot-Script Protections
[x][ =======================================================================
[*] SMRAM: Base = 0x00000000BB000000, Limit = 0x00000000BB7FFFFF, Size = 0x00800000
[!] Found 1 S3 boot-script(s) in EFI variables
[*] Checking S3 boot-script at 0x00000000BA171018
[-] S3 boot-script is not in SMRAM
[*] Reading S3 boot-script from memory..
[*] Decoding S3 boot-script opcodes..
[*] Checking entry-points of Dispatch opcodes..
[-] Dispatch opcode (off 0x014E) with entry-point 0x00000000B9FFB260 > UNPROTECTED
[-] Dispatch opcode (off 0x0F2F) with entry-point 0x00000000B9FFB260 > UNPROTECTED
[-] Dispatch opcode (off 0x1217) with entry-point 0x00000000B9FFB260 > UNPROTECTED
[-] Dispatch opcode (off 0x129A) with entry-point 0x00000000B9FFB260 > UNPROTECTED
[-] Dispatch opcode (off 0x12F2) with entry-point 0x00000000B9FFB260 > UNPROTECTED
[-] Dispatch opcode (off 0x130A) with entry-point 0x00000000B9FFB260 > UNPROTECTED
[-] Dispatch opcode (off 0x1322) with entry-point 0x00000000B9FFB260 > UNPROTECTED
[-] Dispatch opcode (off 0x133A) with entry-point 0x00000000B9FFB260 > UNPROTECTED
[-] Dispatch opcode (off 0x1352) with entry-point 0x00000000B9FFB260 > UNPROTECTED
[-] Dispatch opcode (off 0x136A) with entry-point 0x00000000B9FFB260 > UNPROTECTED
[-] Dispatch opcode (off 0x1382) with entry-point 0x00000000B9FFB260 > UNPROTECTED
[-] Dispatch opcode (off 0x139A) with entry-point 0x00000000B9FFB260 > UNPROTECTED
[-] Dispatch opcode (off 0x13B2) with entry-point 0x00000000B9FFB260 > UNPROTECTED
[-] Dispatch opcode (off 0x13CA) with entry-point 0x00000000B9FFB260 > UNPROTECTED
[-] Dispatch opcode (off 0x13E2) with entry-point 0x00000000B9FFB260 > UNPROTECTED
[-] Dispatch opcode (off 0x13FA) with entry-point 0x00000000B9FFB260 > UNPROTECTED
[-] Dispatch opcode (off 0x1412) with entry-point 0x00000000B9FFB260 > UNPROTECTED
[-] Dispatch opcode (off 0x1E2D) with entry-point 0x00000000B9FFB260 > UNPROTECTED
[-] Dispatch opcode (off 0x20B8) with entry-point 0x00000000B9FFB260 > UNPROTECTED
[-] Dispatch opcode (off 0x24DE) with entry-point 0x00000000B9FFB260 > UNPROTECTED
[-] Dispatch opcode (off 0x24F6) with entry-point 0x00000000B9FFB260 > UNPROTECTED
[-] Dispatch opcode (off 0x26FA) with entry-point 0x00000000B9FFB260 > UNPROTECTED
[-] Dispatch opcode (off 0x2755) with entry-point 0x00000000B9FFB260 > UNPROTECTED
[-] Dispatch opcode (off 0x276D) with entry-point 0x00000000B9FFB260 > UNPROTECTED
[-] Dispatch opcode (off 0x2785) with entry-point 0x00000000B9FFB260 > UNPROTECTED
[-] Dispatch opcode (off 0x279D) with entry-point 0x00000000B9FFB260 > UNPROTECTED
[-] Dispatch opcode (off 0x27B5) with entry-point 0x00000000B9FFB260 > UNPROTECTED
[-] Dispatch opcode (off 0x27CD) with entry-point 0x00000000B9FFB260 > UNPROTECTED
[-] Dispatch opcode (off 0x2C77) with entry-point 0x00000000B9FFB260 > UNPROTECTED
[-] Dispatch opcode (off 0x34C3) with entry-point 0x00000000B9FFB260 > UNPROTECTED
[-] Dispatch opcode (off 0x3523) with entry-point 0x00000000B9FFB260 > UNPROTECTED
[-] Dispatch opcode (off 0x355F) with entry-point 0x00000000B9FFB260 > UNPROTECTED
[-] Dispatch opcode (off 0x3BBB) with entry-point 0x00000000B9FFB260 > UNPROTECTED
[-] Dispatch opcode (off 0x426C) with entry-point 0x00000000B9FFB260 > UNPROTECTED
[-] Dispatch opcode (off 0x431A) with entry-point 0x00000000B9FFB260 > UNPROTECTED
[-] Dispatch opcode (off 0x4332) with entry-point 0x00000000B9FFB260 > UNPROTECTED
[-] Dispatch opcode (off 0x43B3) with entry-point 0x00000000B9FFB260 > UNPROTECTED
[*] Found 37 Dispatch opcodes
[-] Entry-points of Dispatch opcodes in S3 boot-script are not in protected memory

[-] FAILED: S3 Boot-Script and Dispatch entry-points do not appear to be protected

skochinsky, 2017-05-09:
the FAILED messages mean your BIOS implementation may be *vulnerable* but whitelist passing means it is most likely **not infected**. As for KB3172729 is not installing, Microsoft support may be a better place to figure it out.

GreenlightAU, 2017-05-11:
Good to know infection is unlikely - I guess I'll try my luck with Microsoft.  Thanks!

c7zero, 2017-06-09:
Closing this issue

closed 2017-06-09 by c7zero