chipsec/chipsec #212: feature request: add detection of INTEL-SA-00075

skochinsky, 2017-05-03:
reference: https://security-center.intel.com/advisory.aspx?intelid=INTEL-SA-00075&languageid=en-fr

This consists of two parts:
1. determine ME version
2. detect if AMT is supported and enabled/provisioned

1 can be done using using HECI commands. for HECI communication, either the OS driver can be used, or HECI protocol can be reimplemented over PCI to the ME device (like in `intelmetool`). Then the MKHI command `GEN_GET_MKHI_VERSION` can be used to retrieve the firmware version. I have an example [here](https://github.com/skochinsky/me-tools/blob/master/me_util.py) (uses Windows HECI driver).

2 requires some extra commands to get the feature state. I can investigate it if there's interest.

ismaws, 2017-05-03:
tried intelmetool on a couple of xeon-based arch, seems it does not fully support servers. (keeps returning "Hardware unsupported by intelmetool, exiting")
 
any recommended recipe for Xeon/Server detection?

skochinsky, 2017-05-04:
@ismaws well, the official tool is SpsInfo (available in SPS firmware kits from Intel) but I don't know any open-source implementations. you can try adding your device ID to the macros in `intelmetool.h` and see what happens. Anyway, it's probably better discussed over there.

c7zero, 2017-05-09:
Yup, makes sense to add a check. It's better to implement HECI to ME in CHIPSEC, in a separate HAL component. Thanks! Will look into it.