chipsec/chipsec #222: If a module fails both whitelist and the normal mode does that mean there is likely corruption?

SoundandStuff, 2017-05-22:
I ran whitelist and the normal mode and they both gave me the same result that bios_wp failed and a warning on rtclock. What does this mean and what further steps would you recommend?

[CHIPSEC] ***************************  SUMMARY  ***************************
[CHIPSEC] Time elapsed          0.011
[CHIPSEC] Modules total         17
[CHIPSEC] Modules failed to run 0:
[CHIPSEC] Modules passed        8:
[+] PASSED: chipsec.modules.common.bios_kbrd_buffer
[+] PASSED: chipsec.modules.common.bios_ts
[+] PASSED: chipsec.modules.common.smm
[+] PASSED: chipsec.modules.common.spi_desc
[+] PASSED: chipsec.modules.common.spi_fdopss
[+] PASSED: chipsec.modules.common.spi_lock
[+] PASSED: chipsec.modules.memconfig
[+] PASSED: chipsec.modules.remap
[CHIPSEC] Modules failed        1:
[-] FAILED: chipsec.modules.common.bios_wp
[CHIPSEC] Modules with warnings 1:
[!] WARNING: chipsec.modules.common.rtclock
[CHIPSEC] Modules skipped 7:
[*] SKIPPED: chipsec.modules.common.bios_smi
[*] SKIPPED: chipsec.modules.common.ia32cfg
[*] SKIPPED: chipsec.modules.common.smrr
[*] SKIPPED: chipsec.modules.common.secureboot.variables
[*] SKIPPED: chipsec.modules.common.uefi.access_uefispec
[*] SKIPPED: chipsec.modules.common.uefi.s3bootscript
[*] SKIPPED: chipsec.modules.smm_dma

ErikBjorge, 2017-05-22:
Can you provide the complete whitelist and normal mode log files so we can look at the actual settings and firmware module informaiton?

SoundandStuff, 2017-05-26:
Sure does it save the log file automatically or should I redo the tests?

ErikBjorge, 2017-05-26:
You will need to run the test again.  Just use the -l flag and provide a log file name.

abazhaniuk, 2017-06-03:
you have output with entire chipsec_main. if you run specific module use syntax: chipsec_main -m <module name> in case of whitelist it will be for example: python chipsec_main.py -i -n -m tools.uefi.whitelist -a check,../../whitelist/efi-whitelist/json/efi_lenovo.json,../lenovo_t430/lenovo_t430_spi.bin

closed 2017-10-24 by abazhaniuk