chipsec/chipsec #227: Failed installation on Windows via PIP

halsten, 2017-06-04:
Hi,

I followed the instructions in the PDF manual for windows installation using PIP and I ran into a couple of problems. Here is what I did:

`python -m pip install chipsec`

and the output was as follows:

```
Collecting pywin32 (from chipsec)
  Could not find a version that satisfies the requirement pywin32 (from chipsec) (from versions: )
No matching distribution found for pywin32 (from chipsec)
```

So I tried installing `pywin32` first, since it wasn't doing it automatically. Here is what I did:

`python -m pip install pywin32`

and the output was as follows:

```
Collecting pywin32
  Could not find a version that satisfies the requirement pywin32 (from versions: )
No matching distribution found for pywin32
```

It seems for some reason there is a problem pulling `pywin32` using `PIP`, while this might not be a concrete issue with `chipsec`, I thought I should bring it up anyways.

I fixed this issue by just going to `pywin32` [website](https://sourceforge.net/projects/pywin32/files/pywin32/Build%20221/pywin32-221.win32-py2.7.exe/download) instead and downloading the setup file and installing it manually. This seems to have fixed the first issue.

At this point, I referred back to the section in the PDF manual to just install `chipsec` from the source repository instead. I pulled out the repository and followed the instructions to build it. Here is what I did:

`python setup.py install`

and the output was as follows:

```
running install
running bdist_egg
running egg_info
creating chipsec.egg-info
writing requirements to chipsec.egg-info\requires.txt
writing chipsec.egg-info\PKG-INFO
[...]
Using c:\python27\lib\site-packages
Finished processing dependencies for chipsec==1.3.0
```

Finally, `chipsec` was installed, however, I still need to build the drivers on my own. While I know how to do that, I just find it somehow better if there were pre-built binary versions for the drivers for people who do not have the WDK or simply don't know how to compile them. If I can suggest you providing pre-built binary versions as well, it might be a good option.

--halsten

ErikBjorge, 2017-06-07:
Thanks for reporting this bug.

It looks like as of pywin32 version 219 you actually need to use the package name pypiwin32.  We will need to update the setup script to handle this.

We can also look into providing pre-built drivers to remove the need to build the driver and install the DDK.

skochinsky, 2017-07-01:
Just hit this too, which got me thinking. Do we *really* need full pywin32? Most Win32 APIs can be handled via ctypes with a little work.

ErikBjorge, 2017-07-05:
I think for now the quick fix is to update setup.py to use pypiwin32.  We can look into updating the code to remove the dependency in the future.

ErikBjorge, 2018-12-21:
Updated documentation to use pypiwin32 as a dependency.  Removing CHIPSEC install support with pip until we can support Windows and OSX from documentation.  Closing issue.

closed 2018-12-21 by ErikBjorge