chipsec/chipsec #229: windows driver _swsmi incorrect arguments passing

hardboost, 2017-06-06:
```
;  void
;  _swsmi (
;    unsigned int	smi_code_data	// rcx
;    IN   UINT64	rax_value	// rdx
;    IN   UINT64	rbx_value	// r8
;    IN   UINT64	rcx_value	// r9
;    IN   UINT64	rdx_value	// r10
;    IN   UINT64	rsi_value	// r11
;    IN   UINT64	rdi_value	// r12
;    )
```

According to x64 calling convetion first 4 arguments should be passed in registers `rcx, rdx, r8, r9` and remaining arguments should be passed on the stack. Currently only `rax, rbx, rcx` have correct value on the SMI entry and others have some junk in them.

Also i believe there is no need to mess with dx register like this `mov dx, 0B2h`. Isn't it easier to call SMI with `out 0B2h, ax` or even `out 0B2h, al` statement?

ErikBjorge, 2017-06-06:
Thanks for reporting this.  We will look into the calling convention issue this week and should have an update to this issue soon.

BrentHoltsclaw, 2017-06-09:
The issue has been confirmed, we will be working on a fix.

c7zero, 2017-06-09:
Should be fixed by PR #235

ErikBjorge, 2017-06-16:
Closing this issue with fix PR #235.  Please reopen if the problem was not resolved.

closed 2017-06-16 by ErikBjorge