chipsec/chipsec #231: `chipsec_util decode types` causes an error

prashast, 2017-06-06:
I had installed chipsec from the github repo on ubuntu 16.04 and running `chipsec_util decode types` causes an error to occur as follows:
```
################################################################
##                                                            ##
##  CHIPSEC: Platform Hardware Security Assessment Framework  ##
##                                                            ##
################################################################
[CHIPSEC] Version 1.3.0
[CHIPSEC] API mode: using OS native API (not using CHIPSEC kernel module)
[CHIPSEC] Executing command 'decode' with args ['types']

Traceback (most recent call last):
  File "/usr/local/bin/chipsec_util", line 9, in <module>
    load_entry_point('chipsec==1.3.0', 'console_scripts', 'chipsec_util')()
  File "/usr/local/lib/python2.7/dist-packages/chipsec-1.3.0-py2.7-linux-x86_64.egg/chipsec_util.py", line 259, in main
    return chipsecUtil.main()
  File "/usr/local/lib/python2.7/dist-packages/chipsec-1.3.0-py2.7-linux-x86_64.egg/chipsec_util.py", line 226, in main
    comm.run()
  File "/usr/local/lib/python2.7/dist-packages/chipsec-1.3.0-py2.7-linux-x86_64.egg/chipsec/utilcmd/decode_cmd.py", line 71, in run
    print "\n<fw_type> should be in [ %s ]\n" % ( " | ".join( ["%s" % t for t in uefi.fw_types] ) )
AttributeError: 'module' object has no attribute 'fw_types'
```

ErikBjorge, 2017-06-06:
Thanks for reporting this issue.  I was able to reproduce the issue on my system here.  I will debug and provide a pull request.

ErikBjorge, 2017-06-07:
PR #232 should solve this issue (https://github.com/chipsec/chipsec/pull/232)

ErikBjorge, 2017-06-16:
Closing this issue with fix PR #232.  Please reopen if the problem was not resolved.

closed 2017-06-16 by ErikBjorge