chipsec/chipsec #233: chipsec_util smi command line error

abazhaniuk, 2017-06-09:
user@biostraining:~/Desktop/chipsec$ sudo python chipsec_util.py smi 

################################################################
##                                                            ##
##  CHIPSEC: Platform Hardware Security Assessment Framework  ##
##                                                            ##
################################################################
[CHIPSEC] Version 1.3.0
[CHIPSEC] API mode: using OS native API (not using CHIPSEC kernel module)
[CHIPSEC] Executing command 'smi' with args []

Traceback (most recent call last):
  File "chipsec_util.py", line 263, in <module>
    sys.exit( main() )
  File "chipsec_util.py", line 259, in main
    return chipsecUtil.main()
  File "chipsec_util.py", line 226, in main
    comm.run()
  File "/home/user/Desktop/chipsec/chipsec/utilcmd/interrupts_cmd.py", line 57, in run
    op = self.argv[2]
IndexError: list index out of range
user@biostraining:~/Desktop/chipsec$

johnloucaides, 2017-06-09:
please retest and confirm fix :)

ErikBjorge, 2017-06-16:
Closing this issue with fix PR #234.  Please reopen if the problem was not resolved.

closed 2017-06-16 by ErikBjorge