chipsec/chipsec #245: Possibly compromised mini ITX VPN/Firewall?

DorkVader, 2017-06-17:
Hi,
I am a total noob so please bare with me. I got this VPNbox wich was a chinese made miniITX machine model 1037U. It came with pfSense on it and was working really well initially. But soon it started failing. When i plugged a screen into it i could see the bootprocess, but right after instead of a BSD prompt there was alot of random charachters flashing by on the screen. This is when i thought the machine could be compromised. So i got ubuntu running from USB with chipsec on it. so i ran **sudo chipsec_main**. Instead of posting the full log i will post my platform and the summary to start with. Maybe i could get some help from there. If there is anything else from the log you want me to post then please let me know.

################################################################
##                                                            ##
##  CHIPSEC: Platform Hardware Security Assessment Framework  ##
##                                                            ##
################################################################
[CHIPSEC] Version 1.3.1h
[CHIPSEC] Arguments: 
****** Chipsec Linux Kernel module is licensed under GPL 2.0
[CHIPSEC] API mode: using CHIPSEC kernel module API
[CHIPSEC] OS      : Linux 4.8.0-36-generic #36~16.04.1-Ubuntu SMP Sun Feb 5 09:39:57 UTC 2017 x86_64
[CHIPSEC] Platform: Mobile 3rd Generation Core Processor (Ivy Bridge CPU / Panther Point PCH)
[CHIPSEC]      VID: 8086
[CHIPSEC]      DID: 0154

And the summary:

[CHIPSEC] ***************************  SUMMARY  ***************************
[CHIPSEC] Time elapsed          0.434
[CHIPSEC] Modules total         17
[CHIPSEC] Modules failed to run 0:
[CHIPSEC] Modules passed        11:
[+] PASSED: chipsec.modules.common.smrr
[+] PASSED: chipsec.modules.common.bios_kbrd_buffer
[+] PASSED: chipsec.modules.common.bios_ts
[+] PASSED: chipsec.modules.common.smm
[+] PASSED: chipsec.modules.common.rtclock
[+] PASSED: chipsec.modules.common.spi_fdopss
[+] PASSED: chipsec.modules.common.ia32cfg
[+] PASSED: chipsec.modules.common.spi_lock
[+] PASSED: chipsec.modules.remap
[+] PASSED: chipsec.modules.smm_dma
[+] PASSED: chipsec.modules.memconfig
[CHIPSEC] Modules failed        4:
[-] FAILED: chipsec.modules.common.bios_smi
[-] FAILED: chipsec.modules.common.spi_desc
[-] FAILED: chipsec.modules.common.bios_wp
[-] FAILED: chipsec.modules.common.uefi.s3bootscript
[CHIPSEC] Modules with warnings 2:
[!] WARNING: chipsec.modules.common.uefi.access_uefispec
[!] WARNING: chipsec.modules.common.secureboot.variables
[CHIPSEC] Modules skipped 0:
[CHIPSEC]  *****************************************************************

DorkVader, 2017-06-17:
Full output is here:

https://pastebin.com/raw/MgZyUWB3

ErikBjorge, 2017-06-19:
Based on the log file it appears that the SPI flash is not protected but this only indicates potential issues in the firmware implementation.

You may want to run the blacklist tool to see if you have any known bad modules and also generate the hashes for your existing UEFI modules with the whitelist tool.

sudo chipsec_main -m tools.uefi.blacklist
sudo chipsec_main -m tools.uefi.whitelist

The whitelist tool will generate fw.bin and efilist.json.  fw.bin is the SPI flash image and efilist.json contains the hashes for all the UEFI modules.

You may be able to check your hashes against known good values by determining your motherboard vendor and seeing if your hashes match the appropriate whitelist values found here.
https://github.com/advanced-threat-research/efi-whitelist

You can download the appropriate JSON file and compare them using the whitelist tool.  Here is an example command line for checking hashes.
sudo chipsec_main -m tools.uefi.whitelist -a check,efi_intel.json,fw.bin

DorkVader, 2017-06-22:
Thank you for your reply. I will do this later tonight. The machine vendor is a chinese one called Hystou. It was rebranded by a swedish VPN company called OVPN and they sell it as their pfsense based VPNBox. Check it out on ovpn.com if you are curious.

Please keep this thread open as i may have more to add to the conversation.

Thank you!

ErikBjorge, 2018-12-21:
Closing issue.  Please reopen if you have further information.

closed 2018-12-21 by ErikBjorge