chipsec/chipsec #249: Kernel module IOCTL numbers should be generated with <ioctl.h> macros

nickmooney, 2017-06-26:
Best practices for kernel modules dictate that IOCTL numbers should be generated with the _IO, _IOR, _IOW, _IOWR macros rather than laid down sequentially from 0.

This is partially for versioning purposes and partially to ensure that sending an IOCTL to the wrong device does not silently succeed.

nickmooney, 2017-06-26:
Looking at this issue further, it seems the IOCTL numbers *are* defined using the correct macros (in `drivers/linux/include/chipsec.h`), but without a unique parameter given as the first argument to the macro, giving the same result as if they were not used.

BrentHoltsclaw, 2019-03-20:
This should be fixed and merged within #506.  Please reopen the issue if there are farther concerns.  Thanks

closed 2019-03-20 by BrentHoltsclaw