chipsec/chipsec #251: CHIPSEC the Windows driver issue

boxf1, 2017-07-03:
Why delete push rax,rcx,rdx in drivers/win7/amd64/cpu.asm(v1.3.1)?It will cause fuzz test fail.

ErikBjorge, 2017-07-05:
I have started to look into this again.  We will need to check the calling code to see how it handles the registers.  @andrew-furtak may be able to provide some more information about the change.

BrentHoltsclaw, 2017-07-05:
Please see: https://docs.microsoft.com/en-us/cpp/build/caller-callee-saved-registers

boxf1, 2017-07-17:
@andrew-furtak can you help to provide some more information about the change?

andrew-furtak, 2017-07-17:
According to Windows x64 calling convention registers RAX, RCX, RDX "are considered volatile and must be considered destroyed on function calls". There is no need to save them. You should check the link provided by BrentHoltsclaw.

boxf1, 2017-07-18:
@andrew-furtak Our some fuzz test will fail when we don't save them, and the test will pass when we save them.

BrentHoltsclaw, 2017-07-18:
@boxf1 can you provide a verbose log

ErikBjorge, 2017-07-18:
If you are able to provide a verbose log file as @BrentHoltsclaw suggests we may be able to debug the problem you are seeing.  Can we also get a better description of the failure that you are seeing.  We need to be able to determine if the failure is due to the SMI handler updating the buffer or register corruption.

andrew-furtak, 2017-07-18:
My guess is SMI handler may change registers we do not preserve. This may lead to a failure. The solution is to save all the non-volatile registers and restore them back on return.

ErikBjorge, 2017-07-18:
I can update the Windows driver as well as the UEFI Shell code that I just submitted as a pull request to do this.

boxf1, 2017-07-19:
[0x43.txt](https://github.com/chipsec/chipsec/files/1158382/0x43.txt)

ErikBjorge, 2017-07-19:
So this may be a real failure.  Can you please run the same test and also include the -v flag so we can verify the buffer values?

The bug that was fixed in the Windows driver was that RSI and RDI were not getting set correctly on the second and third SMI.  This left random values in these registers when generating the SMI instead of the buffer address.  After the fix the registers are being set correctly to point to the buffer and the test can detect any changes that are made by the SMI.

Now that RSI is being set correctly when generating the SMI the test is detecting the failure.

boxf1, 2017-07-20:
@ErikBjorge Please provide a example of usage for -v.

ErikBjorge, 2017-07-20:
The -v flag is just added to the existing set of parameters to enable verbose output.  The example below assumes that you are running from source.

python chipsec_main.py -v -m tools.smm.smm_ptr -a fuzz,0x43:0x43 -l 0x43.txt

boxf1, 2017-07-21:
[43-v.txt](https://github.com/chipsec/chipsec/files/1164094/43-v.txt)

andrew-furtak, 2017-07-21:
The test failed due to SMI pointer vulnerability present in the firmware. The value from one of the registers holding 0x5A5A5A5A was written to the buffer pointed to by RSI register by the SMI handler. This is exactly what the fuzzer was created for.

boxf1, 2017-07-21:
How to fix this failure?update the Windows driver?

andrew-furtak, 2017-07-21:
update the firmware :)

andrew-furtak, 2017-07-21:
It's not the driver problem. It's a firmware problem.

boxf1, 2017-07-21:
Do you mean it is BIOS firmware problem?But why we can test pass when we add ""push rax  ,pop rax" in _swsmi (drivers\win7\amd64\cpu.asm)?

andrew-furtak, 2017-07-21:
I did not just removed push rax, pop rax. I fixed parameters passing. Before the fix only the first 2 parameters had correct values. Because of this the fuzzer just didn't work and could not find the firmware issue you just found.

ErikBjorge, 2017-07-21:
If you just add a push RAX and a pop RAX in the source you have modified the stack offsets for the parameters that are passed via the stack.  Just adding these instructions without updating the parameter offsets introduces a new bug and RSI will not get set with the correct pointer value.

The change to the driver fixed our generation of SMIs when the pointer is passed in RSI and RDI.  The bug was due to not correctly handling parameter passing.  Because of this fix new failures could be found in system firmware (BIOS).

The log you provided shows the memory modification in the test buffer after generating the particular SMI with the buffer address in RSI.

boxf1, 2017-07-23:
[image]

andrew-furtak, 2017-07-23:
@box1, adding "push rax" you change the stack pointer. The following code getting the parameters from stack gets wrong values. This is why the test "passes".

ErikBjorge, 2017-07-24:
@andrew-furtak is correct.  The change you made broke the code and introduced a bug that keeps RDX, RSI and RDI from getting the correct values from the stack.

closed 2017-08-21 by boxf1