chipsec/chipsec #259: Can't build the chipsec on x86 64-bit GNU/Linux

daedalus, 2017-07-31:
Debian 9 
gcc version 6.3.0 20170406 (Debian 6.3.0-12)
Linux 4.10.0-trunk-amd64 #1 SMP Debian 4.10.7-1~exp1 (2017-03-30) x86_64 GNU/Linux

sudo python setup.py install
/usr/lib/python2.7/dist-packages/setuptools/dist.py:342: UserWarning: The version specified ('1.3.1h') is an invalid version, this may not work as expected with newer versions of setuptools, pip, and PyPI. Please see PEP 440 for more details.
  "details." % self.metadata.version
running install
running bdist_egg
running egg_info
creating chipsec.egg-info
writing chipsec.egg-info/PKG-INFO
writing top-level names to chipsec.egg-info/top_level.txt
writing dependency_links to chipsec.egg-info/dependency_links.txt
writing entry points to chipsec.egg-info/entry_points.txt
writing manifest file 'chipsec.egg-info/SOURCES.txt'
reading manifest file 'chipsec.egg-info/SOURCES.txt'
writing manifest file 'chipsec.egg-info/SOURCES.txt'
installing library code to build/bdist.linux-x86_64/egg
running install_lib
running build_py
creating build
creating build/lib.linux-x86_64-2.7
copying chipsec_main.py -> build/lib.linux-x86_64-2.7
copying chipsec_util.py -> build/lib.linux-x86_64-2.7
creating build/lib.linux-x86_64-2.7/chipsec
copying chipsec/file.py -> build/lib.linux-x86_64-2.7/chipsec
copying chipsec/chipset.py -> build/lib.linux-x86_64-2.7/chipsec
copying chipsec/command.py -> build/lib.linux-x86_64-2.7/chipsec
copying chipsec/module_common.py -> build/lib.linux-x86_64-2.7/chipsec
copying chipsec/defines.py -> build/lib.linux-x86_64-2.7/chipsec
copying chipsec/__init__.py -> build/lib.linux-x86_64-2.7/chipsec
copying chipsec/module.py -> build/lib.linux-x86_64-2.7/chipsec
copying chipsec/logger.py -> build/lib.linux-x86_64-2.7/chipsec
copying chipsec/xmlout.py -> build/lib.linux-x86_64-2.7/chipsec
creating build/lib.linux-x86_64-2.7/chipsec_tools
copying chipsec_tools/__init__.py -> build/lib.linux-x86_64-2.7/chipsec_tools
creating build/lib.linux-x86_64-2.7/chipsec/helper
copying chipsec/helper/oshelper.py -> build/lib.linux-x86_64-2.7/chipsec/helper
copying chipsec/helper/helpers.py -> build/lib.linux-x86_64-2.7/chipsec/helper
copying chipsec/helper/__init__.py -> build/lib.linux-x86_64-2.7/chipsec/helper
creating build/lib.linux-x86_64-2.7/chipsec/cfg
copying chipsec/cfg/common.py -> build/lib.linux-x86_64-2.7/chipsec/cfg
copying chipsec/cfg/__init__.py -> build/lib.linux-x86_64-2.7/chipsec/cfg
creating build/lib.linux-x86_64-2.7/chipsec/hal
copying chipsec/hal/pcidb.py -> build/lib.linux-x86_64-2.7/chipsec/hal
copying chipsec/hal/cpu.py -> build/lib.linux-x86_64-2.7/chipsec/hal
copying chipsec/hal/io.py -> build/lib.linux-x86_64-2.7/chipsec/hal
copying chipsec/hal/igd.py -> build/lib.linux-x86_64-2.7/chipsec/hal
copying chipsec/hal/ucode.py -> build/lib.linux-x86_64-2.7/chipsec/hal
copying chipsec/hal/acpi_tables.py -> build/lib.linux-x86_64-2.7/chipsec/hal
copying chipsec/hal/physmem.py -> build/lib.linux-x86_64-2.7/chipsec/hal
copying chipsec/hal/paging.py -> build/lib.linux-x86_64-2.7/chipsec/hal
copying chipsec/hal/ec.py -> build/lib.linux-x86_64-2.7/chipsec/hal
copying chipsec/hal/uefi_platform.py -> build/lib.linux-x86_64-2.7/chipsec/hal
copying chipsec/hal/iommu.py -> build/lib.linux-x86_64-2.7/chipsec/hal
copying chipsec/hal/spi_uefi.py -> build/lib.linux-x86_64-2.7/chipsec/hal
copying chipsec/hal/tpm12_commands.py -> build/lib.linux-x86_64-2.7/chipsec/hal
copying chipsec/hal/mmio.py -> build/lib.linux-x86_64-2.7/chipsec/hal
copying chipsec/hal/vmm.py -> build/lib.linux-x86_64-2.7/chipsec/hal
copying chipsec/hal/tpm.py -> build/lib.linux-x86_64-2.7/chipsec/hal
copying chipsec/hal/uefi.py -> build/lib.linux-x86_64-2.7/chipsec/hal
copying chipsec/hal/iobar.py -> build/lib.linux-x86_64-2.7/chipsec/hal
copying chipsec/hal/cmos.py -> build/lib.linux-x86_64-2.7/chipsec/hal
copying chipsec/hal/cpuid.py -> build/lib.linux-x86_64-2.7/chipsec/hal
copying chipsec/hal/spd.py -> build/lib.linux-x86_64-2.7/chipsec/hal
copying chipsec/hal/__init__.py -> build/lib.linux-x86_64-2.7/chipsec/hal
copying chipsec/hal/msgbus.py -> build/lib.linux-x86_64-2.7/chipsec/hal
copying chipsec/hal/hal_base.py -> build/lib.linux-x86_64-2.7/chipsec/hal
copying chipsec/hal/spi.py -> build/lib.linux-x86_64-2.7/chipsec/hal
[... 221 lines cut ...]
  File "/usr/lib/python2.7/distutils/dist.py", line 972, in run_command
    cmd_obj.run()
  File "/usr/lib/python2.7/dist-packages/setuptools/command/bdist_egg.py", line 161, in run
    cmd = self.call_command('install_lib', warn_dir=0)
  File "/usr/lib/python2.7/dist-packages/setuptools/command/bdist_egg.py", line 147, in call_command
    self.run_command(cmdname)
  File "/usr/lib/python2.7/distutils/cmd.py", line 326, in run_command
    self.distribution.run_command(command)
  File "/usr/lib/python2.7/distutils/dist.py", line 972, in run_command
    cmd_obj.run()
  File "/usr/lib/python2.7/dist-packages/setuptools/command/install_lib.py", line 24, in run
    self.build()
  File "/usr/lib/python2.7/distutils/command/install_lib.py", line 111, in build
    self.run_command('build_ext')
  File "/usr/lib/python2.7/distutils/cmd.py", line 326, in run_command
    self.distribution.run_command(command)
  File "/usr/lib/python2.7/distutils/dist.py", line 972, in run_command
    cmd_obj.run()
  File "setup.py", line 112, in run
    self._build_linux_driver()
  File "setup.py", line 77, in _build_linux_driver
    subprocess.check_output(["make", "-C", build_driver])
  File "/usr/lib/python2.7/subprocess.py", line 219, in check_output
    raise CalledProcessError(retcode, cmd, output=output)
subprocess.CalledProcessError: Command '['make', '-C', '/home/dclavijo/sync/incoming/code/SEC/chipsec/build/lib.linux-x86_64-2.7/drivers/linux']' returned non-zero exit status 2

abazhaniuk, 2017-07-31:
Could you provide log from: make -C drivers/linux/
Please?

BrentHoltsclaw, 2017-08-15:
Can you please confirm that make is installed?  I was able to create a similar error because I did not have it installed on a Debian build.  If this is the case we will update the documentation to include it within the instructions.

daedalus, 2017-10-30:
its working now

closed 2017-10-30 by daedalus