chipsec/chipsec #262: "Error: cannot copy tree 'drivers/osx': not a directory" on MacOS pip install

ThePuceMoose, 2017-08-09:
When running "pip2 install chipsec" the install fails with message "Error: cannot copy tree 'drivers/osx': not a directory"

A previous issue stated this was because setuptools is not installed, however running "pip2 list" shows setuptools (36.2.7)

Python 2.7.13 was installed through Homebrew, and xcode-select v.2347 is installed.
OS: macOS Sierra v10.12.5

BrentHoltsclaw, 2017-08-15:
The pip installer does not contain the driver/osx directory which is why you are seeing this issue.  Please download the source from github to install.  We are currently working on a fix.

ErikBjorge, 2018-12-21:
We have discontinued pip support for now until we can successfully support all OS versions.  Closing issue.

closed 2018-12-21 by ErikBjorge