chipsec/chipsec #263: Cannot build on macOS with xcode-select

ThePuceMoose, 2017-08-09:
Attempting to run "python setup.py install" returns non-zero exit status error code 1.
"xcode-select: error: tool 'xcodebuild' requires Xcode, but active developer directory '/Library/Developer/CommandLineTools' is a command line tools instance"

It appears xcodebuild is no longer usable on macOS with xcode-select, and requires the full XCode app. Recommend updating the "Mac OS X Installation" section of the manual, as the provided link instructs to install xcode-select.

python 2.7.13
setuptools 36.2.7
xcode-select v2347
macOS Sierra 10.12.5

BrentHoltsclaw, 2017-08-15:
Will the following be sufficient?

> Please  follow the steps below to install CHIPSEC on Mac OS X.
>
>1.  Install XCODE from App Store
>2. Install Python 2.7, PIP and setuptools packages.  Example for installing can be found here: http://docs.python-guide.org/en/latest/starting/install/osx
>3. Turn the System Integrity Protection (SIP) off.  See Configuring SIP

ThePuceMoose, 2017-08-16:
Yes, that's perfect :)

ErikBjorge, 2018-12-21:
Need to update wiki with content.

BrentHoltsclaw, 2019-01-04:
Updated wiki https://github.com/chipsec/chipsec/wiki/Installing-CHIPSEC-in-MacOS and updated instructions within Pull Request #503.  Closing Issue.  Please reopen if anything needs to be added or was missed.

closed 2019-01-04 by BrentHoltsclaw