chipsec/chipsec #266: msgbus on skl

abazhaniuk, 2017-08-15:
Message Bus on skl crashing: 

livecd chipsec-master # python2.7 chipsec_util.py msgbus read 0xb8 0x4

################################################################
##                                                            ##
##  CHIPSEC: Platform Hardware Security Assessment Framework  ##
##                                                            ##
################################################################
[CHIPSEC] Version 1.3.1h
****** Chipsec Linux Kernel module is licensed under GPL 2.0
[CHIPSEC] API mode: using CHIPSEC kernel module API
[CHIPSEC] Executing command 'msgbus' with args ['read', '0xb8', '0x4']

[CHIPSEC] msgbus read: port 0xB8 + 0x00000004
Traceback (most recent call last):
  File "chipsec_util.py", line 262, in <module>
    sys.exit( main() )
  File "chipsec_util.py", line 258, in main
    return chipsecUtil.main()
  File "chipsec_util.py", line 225, in main
    comm.run()
  File "/root/chipsec-master/chipsec/utilcmd/msgbus_cmd.py", line 78, in run
    res = _msgbus.msgbus_reg_read( port, reg )
  File "/root/chipsec-master/chipsec/hal/msgbus.py", line 187, in msgbus_reg_read
    return self.msgbus_read_message(port, register, MessageBusOpcode.MB_OPCODE_REG_READ)
  File "/root/chipsec-master/chipsec/hal/msgbus.py", line 138, in msgbus_read_message
    mcr  = self.__MB_MESSAGE_MCR(port, register, opcode)
  File "/root/chipsec-master/chipsec/hal/msgbus.py", line 110, in __MB_MESSAGE_MCR
    mcr = self.cs.set_register_field( 'MSG_CTRL_REG', mcr, 'MESSAGE_WR_BYTE_ENABLES', 0xF )
  File "/root/chipsec-master/chipsec/chipset.py", line 649, in set_register_field
    field_attrs = self.get_register_def(reg_name)['FIELDS'][field_name]
  File "/root/chipsec-master/chipsec/chipset.py", line 555, in get_register_def
    return self.Cfg.REGISTERS[reg_name]
KeyError: 'MSG_CTRL_REG'
livecd chipsec-master #

ErikBjorge, 2017-08-15:
The interface was updated to include an mm_read for platforms like SKL.  Please verify that chipsec_util shows mm_read as a valid msgbus command.  If so, please give the same register read a try with that version.

****** Chipsec Linux Kernel module is licensed under GPL 2.0
[CHIPSEC] API mode: using CHIPSEC kernel module API
[CHIPSEC] Executing command 'msgbus' with args ['mm_read', '0xb8', '0x4']

[CHIPSEC] MMIO msgbus read: port 0xB8 + 0x00000004
[CHIPSEC] result: 0x00000001
[CHIPSEC] (msgbus) time elapsed 0.000

abazhaniuk, 2017-08-15:
in one system, mm_read works:

livecd chipsec-master # python2.7 chipsec_util.py msgbus mm_read 0xb8 0x4

################################################################
##                                                            ##
##  CHIPSEC: Platform Hardware Security Assessment Framework  ##
##                                                            ##
################################################################
[CHIPSEC] Version 1.3.1h
****** Chipsec Linux Kernel module is licensed under GPL 2.0
[CHIPSEC] API mode: using CHIPSEC kernel module API
[CHIPSEC] Executing command 'msgbus' with args ['mm_read', '0xb8', '0x4']

[CHIPSEC] MMIO msgbus read: port 0xB8 + 0x00000004
[CHIPSEC] result: 0x00000000
[CHIPSEC] (msgbus) time elapsed 0.000
livecd chipsec-master # python2.7 chipsec_util.py msgbus mm_read 0xb8 0x8

################################################################
##                                                            ##
##  CHIPSEC: Platform Hardware Security Assessment Framework  ##
##                                                            ##
################################################################
[CHIPSEC] Version 1.3.1h
****** Chipsec Linux Kernel module is licensed under GPL 2.0
[CHIPSEC] API mode: using CHIPSEC kernel module API
[CHIPSEC] Executing command 'msgbus' with args ['mm_read', '0xb8', '0x8']

[CHIPSEC] MMIO msgbus read: port 0xB8 + 0x00000008
[CHIPSEC] result: 0x04000018
[CHIPSEC] (msgbus) time elapsed 0.000
livecd chipsec-master # 

should we remove  "msgbus read" or it is used in different architecture? 

btw, in other system, mm_read returns all FFs:

user@ubuntudev2:~/chipsec$ sudo python2.7 chipsec_util.py msgbus mm_read 0xB8 0x4

################################################################
##                                                            ##
##  CHIPSEC: Platform Hardware Security Assessment Framework  ##
##                                                            ##
################################################################
[CHIPSEC] Version 1.3.1h
****** Chipsec Linux Kernel module is licensed under GPL 2.0
[CHIPSEC] API mode: using CHIPSEC kernel module API
[CHIPSEC] Executing command 'msgbus' with args ['mm_read', '0xB8', '0x4']

[CHIPSEC] MMIO msgbus read: port 0xB8 + 0x00000004
[CHIPSEC] result: 0xFFFFFFFF
[CHIPSEC] (msgbus) time elapsed 0.002
user@ubuntudev2:~/chipsec$ sudo python2.7 chipsec_util.py msgbus mm_read 0xB8 0x8

################################################################
##                                                            ##
##  CHIPSEC: Platform Hardware Security Assessment Framework  ##
##                                                            ##
################################################################
[CHIPSEC] Version 1.3.1h
****** Chipsec Linux Kernel module is licensed under GPL 2.0
[CHIPSEC] API mode: using CHIPSEC kernel module API
[CHIPSEC] Executing command 'msgbus' with args ['mm_read', '0xB8', '0x8']

[CHIPSEC] MMIO msgbus read: port 0xB8 + 0x00000008
[CHIPSEC] result: 0xFFFFFFFF
[CHIPSEC] (msgbus) time elapsed 0.002
user@ubuntudev2:~/chipsec$

ErikBjorge, 2017-08-15:
The command msgbus read is used for SoC platforms so we need to keep it.  We may need to explain the difference between the two access methods.

Can you attach a verbose log for the system that is returning all FFs?  I would like to see the output from that to see if I can figure out what is going wrong.

abazhaniuk, 2017-08-18:
[msgbus_output.txt](https://github.com/chipsec/chipsec/files/1235497/msgbus_output.txt)
looks like in some system not loading config file correctly: 

[*] Couldn't import chipsec.cfg.skl
No module named skl
[*] Configuration Files:

Btw, maybe we want to detect is it SoC or not and execute different flow, in this case we will just one command and chipsec will run platform specific flow.

ErikBjorge, 2018-12-21:
Closing, please reopen if the error continues.

closed 2018-12-21 by ErikBjorge