chipsec/chipsec #279: exit with only "Killed"

tc8831, 2017-09-04:
I use the a module named "boot_script_table.py" from the blog "http://www.linkedin.com/pulse/chipsec-module-exploits-uefi-boot-script-table-mayur-agnihotri".  My computer is ThinkPad T440s.

when I run with command
"
chipsec_main -m boot_script_table 
"
it shows like this:

################################################################
##                                                            ##
##  CHIPSEC: Platform Hardware Security Assessment Framework  ##
##                                                            ##
################################################################
[CHIPSEC] Version 1.3.0
[CHIPSEC] Arguments: -m boot_script_table
****** Chipsec Linux Kernel module is licensed under GPL 2.0
[CHIPSEC] API mode: using CHIPSEC kernel module API
[CHIPSEC] OS      : Linux 4.12.9-300.fc26.x86_64 #1 SMP Fri Aug 25 13:09:43 UTC 2017 x86_64
[CHIPSEC] Platform: 4th Generation Core Processor (Haswell U/Y)
[CHIPSEC]      VID: 8086
[CHIPSEC]      DID: 0A04
 
[+] loaded chipsec.modules.boot_script_table
[*] running loaded modules ..

[*] running module: chipsec.modules.boot_script_table
[x][ =======================================================================
[x][ Module: UEFI boot script table vulnerability exploit
[x][ =======================================================================
Killed
[tc8831@localhost ~]$ 

So, I debug it using pudb, and find the position of exit, when the  "boot_script_table"  call get_EFI_variable() to get the "AcpiGlobalVariable", it at last call the function fcntl.ioctl at line 293 of [LinuxHelper] helper.py(shown as below), then it print "Killed", evenif I want to just step into the function of fcntl.ioctl. 
#############################################################
line 292: def ioctl(self, nr, args, *mutate_flag):                                        
line 293:         return fcntl.ioctl(self.dev_fh, self._ioctl_base + nr, args)  
#############################################################
the value of args for ioctl is shown with the attached picture file.
self=LinuxHelper
args=array('c', 'F\x00\x00\x00g\xfd\x9f\xaf\x10\xec\x00\x00\x8aH\x00\x00\x9d\x00\x00\x00\xfc       
\x00\x00\x00l\x00\x00\x00\xbf\x00\x00\x00^\x00\x00\x00\xe2\x00\x00\x00,\x00\x00\x00.\x0              
0\x00\x00\x12\x00\x00\x00AcpiGlobalVariable')
nr=13
mutate_flag=()

I wonder the "Killed" is triggered by what and how to solve it. Thank you for your help!

skochinsky, 2017-09-04:
sounds like it's killed by the OS, possibly due to some bug in the driver during the ioctl call. Are there any related messages in dmesg/system log? Are you running any kind of security framework/hardened environment (e.g. PaX/grsec)?

skochinsky, 2017-09-04:
P.S. it seems Fedora is using SELinux. maybe check this guide to see if you can find anything;
https://fedoraproject.org/wiki/SELinux_FAQ

abazhaniuk, 2017-09-05:
also correct way to call s3 boot script module is:
sudo python chipsec_main.py -m common.uefi.s3bootscript

ErikBjorge, 2018-12-21:
Closing, please reopen if additional support is required.

closed 2018-12-21 by ErikBjorge