chipsec/chipsec #283: IOCTL_WRITE_PHYSMEM/IOCTL_WRITE_PHYSMEM don't parse the physical address properly.

msuiche, 2017-09-09:
read_phys_mem() and write_phys_mem() pack the structure as the following, which means the lowPart of the physical address is gonna be first and not second. Just like the way PHYSICAL_ADDRESS structure is ordered.
```
in_buf = struct.pack( '<4IQ', phys_address_lo, phys_address_hi, len(buf), 0, addressof(write_buf) )
```
e.g.:
```
>>> in_buf = pack( '<4IQ', 0x41414141, 0x42424242, 0x43434343, 0, 666)
>>> in_buf
'AAAABBBBCCCC\x00\x00\x00\x00\x9a\x02\x00\x00\x00\x00\x00\x00'
```

And the Windows driver side parse the input buffer, by assuming the first DWORD is the `highPart` of the physical address instead of the `lowPart`.
```
            pInBuf = Irp->AssociatedIrp.SystemBuffer;
            pOutBuf = Irp->AssociatedIrp.SystemBuffer;

            phys_addr.HighPart = ((UINT32*)pInBuf)[0];
            phys_addr.LowPart  = ((UINT32*)pInBuf)[1];
            len                = ((UINT32*)pInBuf)[2];
```

BrentHoltsclaw, 2017-09-11:
It appears that you were looking at the rwehelper and not the win32 helper?  The win32 helper maps to the driver.c code which you reference above.  The code has the following structure:
`in_buf = struct.pack( '3I', phys_address_hi, phys_address_lo, length ) + buf`
The rwehelper is for when you have rwe installed.

ErikBjorge, 2018-12-21:
Closing, please reopen if needed.

closed 2018-12-21 by ErikBjorge