chipsec/chipsec #288: Move from py2exe to pyinstaller

tweksteen, 2017-09-13:
With the end of support for py2exe (https://sourceforge.net/p/py2exe/mailman/message/36033869/), now might be the time to move to pyinstaller for the Windows package. 

I have started a branch where I am trying to clean up the loading of modules. Few questions:
- Do we still want to support the packaged version of Chipsec for Windows (as two standalone .exe)?
- If so, is that a requirement to be able to load extra module (.py) from this packaged version?

ErikBjorge, 2017-09-15:
Switching to pyinstaller seems like a good change.  I do not see a need to be able to load extra modules at this point.  Let's see if anyone has a different opinion.

Slapdog, 2017-09-15:
Should be good idea.  my first few installs ended up crashing the boot near
end of updates. I'm going to start again in about 30 mins, setting up live
boot, and essentials.

npmitche, 2024-01-30:
Removed py2exe. Not supporting.

closed 2024-01-30 by npmitche