chipsec/chipsec #289: SW sequencing verification with Chipsec on older Intel® platforms?

DeniseChien, 2017-09-20:
Dear Sir, 

I tried to validate SW sequencing with Chipsec on older Intel® platforms:
E.g., 4th Generation mobile Intel® Core i7 vPro Processor (Haswell)

Then, SUT encounters below failures:
**[*] Check for presence of certain OPCODEs..
[!] OPCODE detected: OPCODE7 = 0xFF
[!] OPCODE detected: OPCODE6 = 0xFF
[!] OPCODE detected: OPCODE5 = 0xFF
[!] OPCODE detected: OPCODE4 = 0xFF
[!] OPCODE detected: OPCODE1 = 0xD8
[-] FAILED: Bad OPCODEs found!**

Meanwhile, I also tried to use another tool (Intel® Brand Verification Tool) to verify and observed the Security Test can get all PASSED results.

Could you please kindly inform if Chipsec is able to test older products that support SPI SW sequencing?

Thanks for your guidance in advance.

Best regards, 
Denise

ErikBjorge, 2017-09-20:
Can you please provide a verbose log file for the sw_seq test?  This will allow me to debug this issue further.  The test should support Haswell but I need to see a log to see if anything unexpected happened.

ErikBjorge, 2017-09-20:
This test is not present in this project.  Please contact chipsec@intel.com for support.

closed 2017-09-20 by ErikBjorge