chipsec/chipsec #296: BDW.XML update.

ccabobchen, 2017-10-06:
About BDW.XML any update? Because run "sw_seq.py" on Broadwell-U platform always fail. Anybody knows update this file schedule.?

 Thank you for your help!

ErikBjorge, 2017-10-06:
Please send the failing verbose log file to chipsec@intel.com so we can debug the issue.

ccabobchen, 2017-10-11:
HI!
    Attached fail log file.

Thanks

BobChen

ccabobchen, 2017-10-11:
Thanks for sending the log file.  It seems that you may have not used all the correct parameters so we just got a normal chipsec log file.  Please run the following command.

python chipsec_main.py -v -m tools.sw_seq -l sw_seq.log

This should run the tools.sw_seq module and export the results to sw_seq.log.  It will also enable verbose output so we can double check the register definitions being used.

Thanks,
-Erik

ccabobchen, 2017-10-11:
HI! Erik:
              Use command “python chipsec_main.py -v -m tools.sw_seq -l sw_seq.log” still fail and attached the log file.

Thanks
BobChen

From: Bjorge, Erik C [mailto:erik.c.bjorge@intel.com]
Sent: Wednesday, October 11, 2017 8:52 AM
To: Bob.Chen; chipsec
Cc: Author; chipsec/chipsec; chipsec/chipsec
Subject: RE: [chipsec/chipsec] BDW.XML update. (#296)

Thanks for sending the log file.  It seems that you may have not used all the correct parameters so we just got a normal chipsec log file.  Please run the following command.

python chipsec_main.py -v -m tools.sw_seq -l sw_seq.log

This should run the tools.sw_seq module and export the results to sw_seq.log.  It will also enable verbose output so we can double check the register definitions being used.

Thanks,
-Erik

ccabobchen, 2017-10-11:
Hi,

A 0x00 PREFIX is simply undefined.  Treat it as a warning.  As a warning, recommendation is to evaluate its proper operation from a security point-of-view for your system.

Hope this helps.

Thanks,

Aaron

closed 2017-10-11 by ErikBjorge