chipsec/chipsec #300: Missing _io in UEFI StdLib

pietrushnic, 2017-10-13:
When I try `python chipsec_main.py` in UEFI Shell I get:

```
FSOpen: Open '\Efi\StdLib\lib\python.27\json\encoder.pyc' Success
FSOpen: Open '\Efi\StdLib\lib\python.27\zipfile.py' Success
FSOpen: Open '\Efi\StdLib\lib\python.27\' Success
FSOpen: Created new directory entry 'zipfile.pyc'
FSOpen: Open '\Efi\StdLib\lib\python.27\zipfile.pyc' Success
FSOpen: Open '\Efi\StdLib\lib\python.27\shutil.py' Success
FSOpen: Open '\Efi\StdLib\lib\python.27\' Success
FSOpen: Created new directory entry 'shutil.pyc'
FSOpen: Open '\Efi\StdLib\lib\python.27\shutil.pyc' Success
FSOpen: Open '\Efi\StdLib\lib\python.27\io.py' Success
FSOpen: Open '\Efi\StdLib\lib\python.27\' Success
FSOpen: Created new directory entry 'io.pyc'
FSOpen: Open '\Efi\StdLib\lib\python.27\io.pyc' Success
Traceback (most recent call last):
  File "chipsec_main.py", line 50, in <module>
FSOpen: Open '\chipsec\chipsec_main.py' Success
    import zipfile
  File "\Efi\StdLib\lib\python.27/zipfile.py", line 6, in <module>
FSOpen: Open '\Efi\StdLib\lib\python.27\zipfile.py' Success
    import io
  File "\Efi\StdLib\lib\python.27/io.py", line 60, in <module>
FSOpen: Open '\Efi\StdLib\lib\python.27\io.py' Success
    import _io
ImportError: No module named _io
Error: Image at 000CE704000 start failed: Aborted
FS0:\chipsec\>•
```

I'm using tianocore payload on top coreboot, but if I understand correctly this should not matter.

BrentHoltsclaw, 2017-10-18:
Can you try running chipsec within Linux and let us know if it works from there?

kidmetal, 2017-10-26:
Chipsec 1.3.4 cannot run in UEFI shell:
[image]

It works within Ubuntu 17.04.

ErikBjorge, 2017-10-26:
This should be fixed in v1.3.5.  Please let us know if you are still seeing any issues.

pietrushnic, 2017-11-06:
@BrentHoltsclaw v1.3.5 in OS works fine. 

@ErikBjorge Unfortunately under UEFI Python have problems:

```
Python.efi
FSOpen: Open '\Tools\Python.efi' Success
FSOpen: Open '\Tools\Python.efi' Success
FSOpen: Open '\Tools\Python.efi' Success
FSOpen: Open '\Tools\Python.efi' Success
[Security] 3rd party image[0] can be loaded after EndOfDxe: PciRoot(0x0)/Pci(0x11,0x0)/Ata(Primary,Master,0x0)/HD(1,GPT,7A058C80-9E17-483D-A5A4-09915F54A459,0x800,0x100000)/\Tools\Python.efi.
InstallProtocolInterface: 5B1B31A1-9562-11D2-8E3F-00A0C969723B CF15E040
Loading driver at 0x000CEA3D000 EntryPoint=0x000CEA3D440 Python.efi
InstallProtocolInterface: BC62157E-3E33-4FEC-9920-2D3B36D750DF CF15EE98
ProtectUefiImageCommon - 0xCF15E040
  - 0x00000000CEA3D000 - 0x000000000016E3E0                                                                                                                                                                                                   
InstallProtocolInterface: 752F3136-4E16-4FDC-A22A-E5F46812F4CA CFD76478
ImportError: No module named site
Error: Image at 000CEA3D000 start failed: Aborted
FSOpen: Open '.' Success
FSOpen: Open '\Tools' Success
```

I also tried to build Python myself, but unfortunately there are bunch of errors in mainline edk2 and I had to give up for now.

Ecosystem seems to be outdated little bit. Lasty commits to Python in edk2 were in April, `pip install` provide 1.3.0 version, manual target 1.3.0 version. Let me know what would be best way to help you improve that situatio.

BTW I'm trying to enable not supported platform, what path you would suggest to take in that situatio ? I'm thinking about AMD SoC.

pietrushnic, 2017-11-07:
It happen that I copied things to incorrect paths confusing `/boot/efi` with `/boot/efi/EFI`. Python and other components should be in further one.

Despite running Python correctly I didn't confirmed that chipsec works since `python chipsec_main.py` already spent 1.5h on loading:

```
(...)
FSOpen: Open '\EFI\chipsec\chipsec\hal\msr.py' Success
FSOpen: Open '\EFI\chipsec\chipsec\hal\' Success
FSOpen: Created new directory entry 'msr.pyc'
FSOpen: Open '\EFI\chipsec\chipsec\hal\msr.pyc' Success
FSOpen: Open '\EFI\chipsec\chipsec\hal\pci.py' Success
FSOpen: Open '\EFI\chipsec\chipsec\hal\' Success
FSOpen: Created new directory entry 'pci.pyc'
FSOpen: Open '\EFI\chipsec\chipsec\hal\pci.pyc' Success
FSOpen: Open '\EFI\chipsec\chipsec\hal\pcidb.py' Success
```

We leave it in that state and see how it will end up.

BTW It looks like someone tries to port Python3 to EDK2 - you can read about it [here](https://lists.01.org/pipermail/edk2-devel/2017-November/016805.html).

ErikBjorge, 2018-12-21:
Please reopen if this issue persists.

closed 2018-12-21 by ErikBjorge