chipsec/chipsec #303: EfiHelper' object has no attribute 'decompress_file

EliseJanezp, 2017-10-16:
################################################################
##                                                            ##
##  CHIPSEC: Platform Hardware Security Assessment Framework  ##
##                                                            ##
################################################################
[CHIPSEC] Version 1.3.3h
[CHIPSEC] Arguments: -i -m tools.uefi.blacklist -l blacklist.txt
[CHIPSEC] API mode: using CHIPSEC kernel module API
[CHIPSEC] OS      : uefi   
[CHIPSEC] Platform: Mobile 5th Generation Core Processor (Broadwell M/H / Wildcat Point PCH)
[CHIPSEC]      VID: 8086
[CHIPSEC]      DID: 1604
 
[+] loaded chipsec.modules.tools.uefi.blacklist
[*] running loaded modules ..

[*] running module: chipsec.modules.tools.uefi.blacklist
[x][ =======================================================================
[x][ Module: Check for black-listed EFI binaries in UEFI firmware
[x][ =======================================================================

Usage:

    chipsec_main.py -i -m tools.uefi.blacklist [-a <fw_image>,<blacklist>]

      fw_image  : Full file path to UEFI firmware image
                  If not specified, the module will dump firmware image directly from ROM
      blacklist : JSON file with configuration of black-listed EFI binaries (default = blacklist.json)
                  Config file should be located in the same directory as this module
    
Examples:

    chipsec_main.py -m tools.uefi.blacklist 

      Dumps UEFI firmware image from flash memory device, decodes it and
      checks for black-listed EFI modules defined in the default config 'blacklist.json'

    chipsec_main.py -i --no_driver -m tools.uefi.blacklist -a uefi.rom,blacklist.json

      Decodes 'uefi.rom' binary with UEFI firmware image and
      checks for black-listed EFI modules defined in 'blacklist.json' config

Important! This module can only detect what it knows about from its config file.
If a bad or vulnerable binary is not detected then its 'signature' needs to be added to the config.

[*] dumping FW image from ROM to fw.bin: 0x00500000 bytes at [0x00FFFFFF:0x00B00000]
[*] this may take a few minutes (instead, use 'chipsec_util spi dump')...
[*] searching for EFI binaries that match criteria from 'blacklist.json':
    Computrace       - Computrace
    HT_UEFI_Rootkit  - HackingTeam UEFI Rootkit (http://www.intelsecurity.com/advanced-threat-research/content/data/HT-UEFI-rootkit.html)
    SmmComputrace    - SmmComputrace
    LenovoComputraceSmiServices - Computrace
    LenovoComputraceLoaderDxe - Computrace
    ThinkPwn         - ThinkPwn: SystemSmmRuntimeRt SMM vulnerability (http://blog.cr4.sh/2016/06/exploring-and-exploiting-lenovo.html)
    LenovoComputraceEnablerDxe - Computrace
    BdsComputrace    - BdsComputrace
    AbsoluteComputraceInstallerWin8 - Computrace
ERROR: Exception occurred during chipsec.modules.tools.uefi.blacklist.run(): ''EfiHelper' object has no attribute 'decompress_file''

[CHIPSEC] ***************************  SUMMARY  ***************************
[CHIPSEC] Time elapsed          46.000
[CHIPSEC] Modules total         1
[CHIPSEC] Modules failed to run 1:
ERROR: chipsec.modules.tools.uefi.blacklist
[CHIPSEC] Modules passed        0:
[CHIPSEC] Modules failed        0:
[CHIPSEC] Modules with warnings 0:
[CHIPSEC] Modules skipped 0:
[CHIPSEC] Modules with Exceptions 1:
ERROR: chipsec.modules.tools.uefi.blacklist
[CHIPSEC] *****************************************************************

BrentHoltsclaw, 2017-10-18:
Decompress functionality was not implemented within the UEFI driver.  As a workaround you can dump the spi within UEFI and then decompress within Linux or Windows.

EliseJanezp, 2017-10-18:
Thanks for your reply,Now I am working under UEFI environment to be wanted to decompress the fuctionality,how long do you complete the function within the UEFI driver.

frinzell, 2021-02-22:
Decompress functionality is not planned for UEFI Shell.  Closing.

closed 2021-02-22 by frinzell