chipsec/chipsec #304: [-] FAILED: chipsec.modules.smm_dma on Atom(CHT)

kmasamune, 2017-10-18:
The current TSEG base is the configurations of Big Core (ex. 7th, 8th processor), and fails at mismatch on Atom(CHT).
Therefore it is not FAILED, it is SKIPPED. SKIPPED is not done automatically, Should I make a SKIPPED decision manually?

johnloucaides, 2017-10-19:
Currently smm_dma is supposed to be SKIPPED on Atom...
`    def is_supported(self):
        # @TODO: currently, this module cannot run on macOS
        if self.cs.helper.is_macos(): return False
        if self.cs.is_atom(): return False
        if self.cs.is_server(): return False
        else: return True`

Can you include a verbose logfile (run with "-v") from a run where it runs on CHT and returns failed? That might help track down why you're not getting SKIPPED.

kmasamune, 2017-10-19:
It is Log file. Two Failed, one is a known issue. It has already been solved.(chipsec.modules.common.bios_wp)

[helper] OS: Windows 8 6.2.9200
[helper] Using 'helper/win/win7_amd64' path for driver
################################################################
##                                                            ##
##  CHIPSEC: Platform Hardware Security Assessment Framework  ##
##                                                            ##
################################################################
[CHIPSEC] Version 1.3.2
[CHIPSEC] Arguments: -v -l LAPIS64-V2.01.txt

WARNING: *******************************************************************
WARNING: Chipsec should only be used on test systems!
WARNING: It should not be installed/deployed on production end-user systems.
WARNING: See WARNING.txt
WARNING: *******************************************************************

[helper] service control manager opened (handle = 0x006d9270)
[helper] driver path: 'D:\CHIPSEC-Rev1_3_2\chipsec\helper\win\win7_amd64\chipsec_hlpr.sys'
[helper] service 'chipsec' created (handle = 0x006d91d0)
[helper] service 'chipsec' started
[CHIPSEC] API mode: using CHIPSEC kernel module API
[helper] opened device '\\.\chipsec_hlpr' (handle: 000003cc)
00 00 00 00 00 00 00 00 04                      |          
[pci] reading B/D/F: 0/0/0, offset: 0x00, value: 0x22808086
[*] Couldn't import chipsec.cfg.cht
No module named cht
[*] Configuration Files:
[*] - D:\CHIPSEC-Rev1_3_2\chipsec/cfg\avn.xml
[*] - D:\CHIPSEC-Rev1_3_2\chipsec/cfg\bdw.xml
[*] - D:\CHIPSEC-Rev1_3_2\chipsec/cfg\bdx.xml
[*] - D:\CHIPSEC-Rev1_3_2\chipsec/cfg\byt.xml
[*] - D:\CHIPSEC-Rev1_3_2\chipsec/cfg\cht.xml
[*] - D:\CHIPSEC-Rev1_3_2\chipsec/cfg\common.xml
[*] - D:\CHIPSEC-Rev1_3_2\chipsec/cfg\hsw.xml
[*] - D:\CHIPSEC-Rev1_3_2\chipsec/cfg\hsx.xml
[*] - D:\CHIPSEC-Rev1_3_2\chipsec/cfg\iommu.xml
[*] - D:\CHIPSEC-Rev1_3_2\chipsec/cfg\ivt.xml
[*] - D:\CHIPSEC-Rev1_3_2\chipsec/cfg\jkt.xml
[*] - D:\CHIPSEC-Rev1_3_2\chipsec/cfg\kbl.xml
[*] - D:\CHIPSEC-Rev1_3_2\chipsec/cfg\kbr.xml
[*] - D:\CHIPSEC-Rev1_3_2\chipsec/cfg\qrk.xml
[*] - D:\CHIPSEC-Rev1_3_2\chipsec/cfg\skl.xml
[*] - D:\CHIPSEC-Rev1_3_2\chipsec/cfg\template.xml
[*] Loading Configuration Files:
[*] looking for platform config in 'D:\CHIPSEC-Rev1_3_2\chipsec/cfg\common.xml'..
[*] loading common platform config from 'D:\CHIPSEC-Rev1_3_2\chipsec/cfg\common.xml'..
[*] loading integrated devices/controllers..
    + HOSTCTRL        : {'fun': '0', 'bus': '0', 'dev': '0x00', 'vid': '0x8086'}
    + PEG10           : {'fun': '0', 'bus': '0', 'dev': '0x01', 'vid': '0x8086'}
    + PEG11           : {'fun': '1', 'bus': '0', 'dev': '0x01', 'vid': '0x8086'}
    + PEG12           : {'fun': '2', 'bus': '0', 'dev': '0x01', 'vid': '0x8086'}
    + IGD             : {'fun': '0', 'bus': '0', 'dev': '0x02', 'vid': '0x8086'}
    + HDAUDIO         : {'fun': '0', 'bus': '0', 'dev': '0x03', 'vid': '0x8086'}
    + XHCI            : {'fun': '0', 'bus': '0', 'dev': '0x14', 'vid': '0x8086'}
    + MEI1            : {'fun': '0', 'bus': '0', 'dev': '0x16', 'vid': '0x8086'}
    + MEI2            : {'fun': '1', 'bus': '0', 'dev': '0x16', 'vid': '0x8086'}
    + IDER            : {'fun': '2', 'bus': '0', 'dev': '0x16', 'vid': '0x8086'}
    + KT              : {'fun': '3', 'bus': '0', 'dev': '0x16', 'vid': '0x8086'}
    + GBE             : {'fun': '0', 'bus': '0', 'dev': '0x19', 'vid': '0x8086'}
    + EHCI2           : {'fun': '0', 'bus': '0', 'dev': '0x1A', 'vid': '0x8086'}
    + AUDIO           : {'fun': '0', 'bus': '0', 'dev': '0x1B', 'vid': '0x8086'}
    + PCIE1           : {'fun': '0', 'bus': '0', 'dev': '0x1C', 'vid': '0x8086'}
    + PCIE2           : {'fun': '1', 'bus': '0', 'dev': '0x1C', 'vid': '0x8086'}
    + PCIE3           : {'fun': '2', 'bus': '0', 'dev': '0x1C', 'vid': '0x8086'}
    + PCIE4           : {'fun': '3', 'bus': '0', 'dev': '0x1C', 'vid': '0x8086'}
    + PCIE5           : {'fun': '4', 'bus': '0', 'dev': '0x1C', 'vid': '0x8086'}
    + PCIE6           : {'fun': '5', 'bus': '0', 'dev': '0x1C', 'vid': '0x8086'}
    + PCIE7           : {'fun': '6', 'bus': '0', 'dev': '0x1C', 'vid': '0x8086'}
[... 892 lines cut ...]
[+] PASSED: chipsec.modules.common.bios_kbrd_buffer
[+] PASSED: chipsec.modules.common.bios_smi
[+] PASSED: chipsec.modules.common.bios_ts
[+] PASSED: chipsec.modules.common.ia32cfg
[+] PASSED: chipsec.modules.common.smrr
[+] PASSED: chipsec.modules.common.spi_desc
[+] PASSED: chipsec.modules.common.spi_fdopss
[+] PASSED: chipsec.modules.common.spi_lock
[+] PASSED: chipsec.modules.common.secureboot.variables
[+] PASSED: chipsec.modules.common.uefi.access_uefispec
[CHIPSEC] Modules failed        2:
[-] FAILED: chipsec.modules.common.bios_wp
[-] FAILED: chipsec.modules.smm_dma
[CHIPSEC] Modules with warnings 1:
[!] WARNING: chipsec.modules.common.uefi.s3bootscript
[CHIPSEC] Modules skipped 4:
[*] SKIPPED: chipsec.modules.common.rtclock
[*] SKIPPED: chipsec.modules.common.smm
[*] SKIPPED: chipsec.modules.memconfig
[*] SKIPPED: chipsec.modules.remap
[CHIPSEC] *****************************************************************
[helper] stopping service 'chipsec'..
[helper] service 'chipsec' stopped
[helper] deleting service 'chipsec'...
[helper] service 'chipsec' deleted

npmitche, 2024-01-30:
Should be fixed since https://github.com/chipsec/chipsec/releases/tag/1.5.0

closed 2024-01-30 by npmitche