chipsec/chipsec #305: How to get the hex offset of the variables from the chipsec report?

xcookie910, 2017-10-18:
Sorry I am a total newbie to chipsec, I would like to flip some bits on a few variables to enable some lockdown features. I noticed on the CHIPSEC report this for example:

> [*] BC = 0x00 << BIOS Control (b:d.f 00:31.0 + 0xDC)    
    [00] BIOSWE           = 0 << BIOS Write Enable     
    [01] BLE              = 0 << BIOS Lock Enable     
    [02] SRC              = 0 << SPI Read Configuration     
    [04] TSS              = 0 << Top Swap Status     
    [05] SMM_BWP          = 0 << SMM BIOS Write Protection     
[-] BIOS region write protection is disabled!   

Okay so it looks like I should flip the 1st and 5th bit to 1 to enable the protection here.

However I can't figure out from this information the offset of the variables in a HEX editor.

I have both the full raw BIOS file and I can also dump the BIOS with chipsec (however it doesn't dump all of it, it leaves out a header of like 170kb size compared to the full BIOS).

So what is the offset of the variables in this case either in the dumped file or in the full file, is the "(b:d.f 00:31.0 + 0xDC)" stuff supposed to represent something? Please help me get the hex offset so that I can edit it in a hex editor.

c7zero, 2017-10-18:
Yeah, the (b:d.f 00:31.0 + 0xDC) is the location of *hardware* register "BIOS Control" containing the bits you want to flip. It's neither in the BIOS dump nor in any variables (hopefully). This particular register is at the offset 0xDC in the PCIe configuration space of the device b:d.f 00:31.0 on the system. In order to modify bits in this register, use "chipsec_util pci" command.

yuriy

xcookie910, 2017-10-24:
@c7zero 

Thank you, after some trial and error I have succesfully figured out how to use the `chipsec_util` tools, it's really easy to use looking back, it is a very useful tool thanks for all your hard work.

I have successfully flipped some bits in the PCI and MMIO parts where some things were not write protected, but the problem is that they are temporary.

After I restarted the computer, everything was set back to default.

I guess this is because chipsec only edits the values in the memory, I guess the firmware is loaded into the memory at every bootup, and chipsec edits that. After the next boot the original firmware is loaded again, overwriting anything that we saved with chipsec.

I have grabbed the firmware file, and saw no bitwise change (sha checksum) compared to the original one from the manufacturer.

How to edit the values in the firmware file itself? Since if we just edit it in the memory that is not very useful if it gets reset after every boot.

--------------

I tried searching for those hex values, for example the MMIO register, the one named `HSFS` group located at `0x04` that is 4 bytes from the `SPIBAR`, the variable named `FLOCKDN` which should be bit 15.

In the memory it was easy to find `3F006008` which to binary it translates to 0011 1111 0000 0000 **0**110 0000 0000 1000 (the highlighted zero value is the FLOCKDN)

For some reason it is read backwards, so the 15th byte was the `0` before the two `11`. So that is easy to lock just change the `3F006008` to `3F00E008` , and it worked it did flipped the FLOCKDN to 1.

Now I tried searching for the same value in the firmware file, and no luck, I guess the SPIBAR is represented by 16 bytes of `F` values and tried counting from there but nowhere I could find the `3F006008` structure.

Is it possible that the firmware file is encoded in some different way?

skochinsky, 2017-10-25:
You should bug your vendor to fix the firmware. You can try to track down where it's done yourself but it's a pretty complicated process, because there are countless ways the code may be accessing those registers.

skochinsky, 2017-10-26:
There are countless ways the firmware may be writing to those registers. You should bug your vendor to fix the firmware so it sets up the protections properly instead of doing it yourself, you'll only lose your time for nothing.

xcookie910, 2017-10-26:
@skochinsky well that would be unproductive, they have stopped the firmware support for this product years ago.

I think I found the region for it though, I don't think the values are randomly spread out, they may be just compressed and show different values.

For example for another value I have found the location to be exactly as what Chipsec reported, SPIBAR+ x for example. And it was exactly there, I just had to find the "SPIBAR" marker and count from that. 

Now the bit values I have trouble with, since that is very weirdly arranged sometimes it starts from backwards sometimes from forward, so I have to test every single combination.

I would be a long work, I was just wondering whether there is a shorter way?

ErikBjorge, 2018-05-21:
If you need more support please reopen this issue.

closed 2018-05-21 by ErikBjorge