chipsec/chipsec #308: Cannot install chipsec using "python setup.py install" on Linux

weitang730, 2017-10-24:
I have an error when running "python setup.py install":
_running install
running bdist_egg
running egg_info
writing chipsec.egg-info/PKG-INFO
writing top-level names to chipsec.egg-info/top_level.txt
writing dependency_links to chipsec.egg-info/dependency_links.txt
writing entry points to chipsec.egg-info/entry_points.txt
reading manifest file 'chipsec.egg-info/SOURCES.txt'
writing manifest file 'chipsec.egg-info/SOURCES.txt'
installing library code to build/bdist.linux-x86_64/egg
running install_lib
running build_py
running build_ext
building the linux driver
make: *** [check_kernel_dir] Error 1
Traceback (most recent call last):
  File "setup.py", line 242, in <module>
    **extra_kw
  File "/usr/lib64/python2.7/distutils/core.py", line 152, in setup
    dist.run_commands()
  File "/usr/lib64/python2.7/distutils/dist.py", line 953, in run_commands
    self.run_command(cmd)
  File "/usr/lib64/python2.7/distutils/dist.py", line 972, in run_command
    cmd_obj.run()
  File "/usr/lib/python2.7/site-packages/setuptools/command/install.py", line 73, in run
    self.do_egg_install()
  File "/usr/lib/python2.7/site-packages/setuptools/command/install.py", line 93, in do_egg_install
    self.run_command('bdist_egg')
  File "/usr/lib64/python2.7/distutils/cmd.py", line 326, in run_command
    self.distribution.run_command(command)
  File "/usr/lib64/python2.7/distutils/dist.py", line 972, in run_command
    cmd_obj.run()
  File "/usr/lib/python2.7/site-packages/setuptools/command/bdist_egg.py", line 185, in run
    cmd = self.call_command('install_lib', warn_dir=0)
  File "/usr/lib/python2.7/site-packages/setuptools/command/bdist_egg.py", line 171, in call_command
    self.run_command(cmdname)
  File "/usr/lib64/python2.7/distutils/cmd.py", line 326, in run_command
    self.distribution.run_command(command)
  File "/usr/lib64/python2.7/distutils/dist.py", line 972, in run_command
    cmd_obj.run()
  File "/usr/lib/python2.7/site-packages/setuptools/command/install_lib.py", line 20, in run
    self.build()
  File "/usr/lib64/python2.7/distutils/command/install_lib.py", line 111, in build
    self.run_command('build_ext')
  File "/usr/lib64/python2.7/distutils/cmd.py", line 326, in run_command
    self.distribution.run_command(command)
  File "/usr/lib64/python2.7/distutils/dist.py", line 972, in run_command
    cmd_obj.run()
  File "setup.py", line 112, in run
    self._build_linux_driver()
  File "setup.py", line 77, in _build_linux_driver
    subprocess.check_output(["make", "-C", build_driver])
  File "/usr/lib64/python2.7/subprocess.py", line 575, in check_output
    raise CalledProcessError(retcode, cmd, output=output)
subprocess.CalledProcessError: Command '['make', '-C', '/root/tools/chipsec-1.3.4-p0006/build/lib.linux-x86_64-2.7/drivers/linux']' returned non-zero exit status 2_

When I run "yum install kernel kernel-devel-$(uname -r)":
_[root@sca05-0a81e17a chipsec-1.3.4-p0006]# yum install kernel kernel-devel-$(uname -r)
Loaded plugins: aliases, changelog, kabi, langpacks, ovl, tmprepo, ulninfo,
              : verify, versionlock
Loading support for kernel ABI
Package kernel-3.10.0-693.5.2.el7.x86_64 already installed and latest version
No package kernel-devel-4.1.12-103.6.1.el7uek.x86_64 available.
Nothing to do_

Is that expected?

chipsecintel, 2017-10-24:
This is not expected. The goal is to get the kernel and kernel-devel packages installed. If you are on a live image, this is a bit tricky. Maybe try:

yum install kernel kernel-devel

Or possibly:

yum install kernel-$(uname -r) kenrel-devel-$(uname -r)

weitang730, 2017-10-24:
Tried but no such package available
[root@sca05-0a81e17a chipsec-1.3.4-p0006]# yum install kernel-$(uname -r) kenrel-devel-$(uname -r)
Loaded plugins: aliases, changelog, kabi, langpacks, ovl, tmprepo, ulninfo,
              : verify, versionlock
Loading support for kernel ABI
No package kernel-4.1.12-103.6.1.el7uek.x86_64 available.
No package kenrel-devel-4.1.12-103.6.1.el7uek.x86_64 available.
Error: Nothing to do

chipsecintel, 2017-10-25:
I guess I should have asked which distro you are using. I only noticed the el7 and assumed RHEL or CentOS. You seem to be on Oracle, and that’s a first as far as I know. Ultimately, you need to be able to build kernel modules. I’ve done that on a bunch of systems, but there are always more. ;-)

Based on google searches, it seems like they call their kernel “kernel-uek” and the development package appears to be “kernel-uek-devel”. Based on this, I guess you should try “yum install kernel-uek kernel-uek-devel”.

Let me know how it goes. ☺

weitang730, 2017-10-25:
Yes, I am on Oracle. 
uname -r : 4.1.12-103.6.1.el7uek.x86_64
After  “yum install kernel-uek kernel-uek-devel”, I installed them successfully.
But the same error occurs again when installing chipsec.

weitang730, 2017-11-16:
After reboot, I can run install and run chipsec successfully.
Thanks,
-Wei

closed 2018-05-21 by ErikBjorge