chipsec/chipsec #309: tools.uefi.uefivar_fuzz fail under LINUX Ubuntu OS

zwangwangwang, 2017-10-24:
It occured a bad problem which test  by tools.uefi.uefivar_fuzz with an error bad address .
There display as follow when use dmseg command in LIUNX SHELL:
 [chipsec] ERROR: STATUS_UNSUCCESSFUL - could not allocate memory

ErikBjorge, 2017-10-25:
Can you please provide the version of Ubuntu that you are using and the output from the command?

zwangwangwang, 2017-10-26:
verison: ubuntu-16.04.2-desktop-amd64

zwangwangwang, 2017-10-26:
HIPSEC: Platform Hardware Security Assessment Framework  ##
##                                                            ##
################################################################
[CHIPSEC] Version 1.3.3h
[CHIPSEC] Arguments: -m tools.uefi.uefivar_fuzz
****** Chipsec Linux Kernel module is licensed under GPL 2.0
[CHIPSEC] API mode: using CHIPSEC kernel module API
[CHIPSEC] OS      : Linux 4.8.0-36-generic #36~16.04.1-Ubuntu SMP Sun Feb 5 09:39:57 UTC 2017 x86_64
[CHIPSEC] Platform: Mobile 5th Generation Core Processor (Broadwell M/H / Wildcat Point PCH)
[CHIPSEC]      VID: 8086
[CHIPSEC]      DID: 1604
 
[+] loaded chipsec.modules.tools.uefi.uefivar_fuzz
[*] running loaded modules ..

[*] running module: chipsec.modules.tools.uefi.uefivar_fuzz
[x][ =======================================================================
[x][ Module: Fuzz UEFI Variable Interface
[x][ =======================================================================
WARNING: Are you sure you want to continue fuzzing UEFI variable interface?
Type 'yes' to continue > yes
Test      : all
Iterations: 1000
Seed      : 1508984214
Test case : 1

  Running test #1:
ERROR: Exception occurred during chipsec.modules.tools.uefi.uefivar_fuzz.run(): '[Errno 14] Bad address'

[CHIPSEC] ***************************  SUMMARY  ***************************
[CHIPSEC] Time elapsed          26.540
[CHIPSEC] Modules total         1
[CHIPSEC] Modules failed to run 1:
ERROR: chipsec.modules.tools.uefi.uefivar_fuzz
[CHIPSEC] Modules passed        0:
[CHIPSEC] Modules failed        0:
[CHIPSEC] Modules with warnings 0:
[CHIPSEC] Modules skipped 0:
[CHIPSEC] Modules with Exceptions 1:
ERROR: chipsec.modules.tools.uefi.uefivar_fuzz
[CHIPSEC] *****************************************************************

zwangwangwang, 2017-10-26:
psec] > IOCTL_SET_EFIVAR
[  224.518164] ------------[ cut here ]------------
[  224.518168] WARNING: CPU: 2 PID: 2496 at /build/linux-hwe-eyfT8D/linux-hwe-4.8.0/mm/page_alloc.c:3421 __alloc_pages_slowpath+0x70c/0xa20
[  224.518168] Modules linked in: chipsec(OE) ccm rfcomm bnep nls_iso8859_1 arc4 iwlmvm mac80211 uvcvideo videobuf2_vmalloc videobuf2_memops intel_rapl videobuf2_v4l2 videobuf2_core x86_pkg_temp_thermal videodev coretemp media kvm_intel snd_hda_codec_hdmi kvm snd_hda_codec_realtek thinkpad_acpi snd_hda_codec_generic snd_hda_intel snd_hda_codec snd_hda_core iwlwifi snd_hwdep btusb btrtl btbcm snd_pcm nvram btintel irqbypass snd_seq_midi snd_seq_midi_event snd_rawmidi snd_seq crct10dif_pclmul crc32_pclmul ghash_clmulni_intel snd_seq_device bluetooth cryptd joydev input_leds intel_cstate intel_rapl_perf serio_raw cfg80211 snd_timer snd soundcore rtsx_pci_ms memstick shpchp mei_me mei lpc_ich intel_pch_thermal mac_hid parport_pc ppdev lp parport autofs4 hid_generic usbhid hid uas usb_storage rtsx_pci_sdmmc
[  224.518196]  nouveau i915 psmouse mxm_wmi ahci ttm libahci i2c_algo_bit drm_kms_helper rtsx_pci syscopyarea sysfillrect sysimgblt fb_sys_fops video drm fjes wmi e1000e ptp pps_core
[  224.518205] CPU: 2 PID: 2496 Comm: python Tainted: G           OE   4.8.0-36-generic #36~16.04.1-Ubuntu
[  224.518205] Hardware name: LENOVO 20BVA017CD/20BVA017CD, BIOS JBET51WW (1.16 ) 07/08/2015
[  224.518207]  0000000000000286 0000000027279118 ffff9ed8c2437aa0 ffffffffa022d7b3
[  224.518209]  0000000000000000 0000000000000000 ffff9ed8c2437ae0 ffffffff9fe8313b
[  224.518210]  00000d5d02000200 00000000024000c0 000000000240c0c0 000000000000000e
[  224.518212] Call Trace:
[  224.518215]  [<ffffffffa022d7b3>] dump_stack+0x63/0x90
[  224.518217]  [<ffffffff9fe8313b>] __warn+0xcb/0xf0
[  224.518218]  [<ffffffff9fe8326d>] warn_slowpath_null+0x1d/0x20
[  224.518219]  [<ffffffff9ffaad7c>] __alloc_pages_slowpath+0x70c/0xa20
[  224.518220]  [<ffffffff9ffa9bfd>] ? get_page_from_freelist+0x42d/0xad0
[  224.518222]  [<ffffffff9feb553c>] ? __enqueue_entity+0x6c/0x70
[  224.518223]  [<ffffffff9ffab34a>] __alloc_pages_nodemask+0x2ba/0x300
[  224.518225]  [<ffffffff9fffec65>] alloc_pages_current+0x95/0x140
[  224.518227]  [<ffffffff9ffcbf9e>] kmalloc_order_trace+0x2e/0xe0
[  224.518228]  [<ffffffffa000bfdd>] __kmalloc+0x1ed/0x200
[  224.518231]  [<ffffffffc0b85251>] d_ioctl+0x8c1/0x1490 [chipsec]
[  224.518232]  [<ffffffff9ffb205d>] ? __pagevec_lru_add_fn+0x12d/0x220
[  224.518234]  [<ffffffff9ffb2e8d>] ? __lru_cache_add+0x6d/0x90
[  224.518235]  [<ffffffff9ffb3796>] ? lru_cache_add_active_or_unevictable+0x36/0xb0
[  224.518237]  [<ffffffff9ffdb6c6>] ? handle_mm_fault+0xf86/0x1410
[  224.518238]  [<ffffffffa0047861>] do_vfs_ioctl+0xa1/0x5f0
[  224.518240]  [<ffffffff9fe6b4f5>] ? __do_page_fault+0x265/0x4e0
[  224.518241]  [<ffffffffa0047e29>] SyS_ioctl+0x79/0x90
[  224.518243]  [<ffffffffa0696576>] entry_SYSCALL_64_fastpath+0x1e/0xa8
[  224.518244] ---[ end trace 04a9c45c4aaa019e ]---
[  224.518245] [chipsec] ERROR: STATUS_UNSUCCESSFUL - could not allocate memory
[  224.525621] chipsec cleanup_module 1772: Destroying chipsec device
[  224.525787] chipsec cleanup_module 1774: exit

ErikBjorge, 2017-11-01:
I was able to reproduce the error on my Ubuntu system as well.  We will dig into the code to see what may be causing the problem.

Thanks for providing the log information.

BrentHoltsclaw, 2019-01-02:
UEFI Variable fuzzer is fixed within the most recent version of Chipsec.  The test was failing because the kernel could not allocate a large enough buffer to fit in the size of the name and data for the variable being generated.

If you feel this is still an issue or would like farther changes please reopen the issue.

closed 2019-01-02 by BrentHoltsclaw