chipsec/chipsec #319: Possible infection?

Folkerttt, 2017-11-07:
Here the output of chipsec_main on Gentoo with Asus uefi/legacy motherboard. 

Failed: chipsec.modules.common.bios_wp
Warning: chipsec. modules.common.spi_acces
Warning: chipsec.modules.common.uefi.s3bootscript

Is the tested machine infected? 

****** Chipsec Linux Kernel module is licensed under GPL 2.0
[CHIPSEC] API mode: using CHIPSEC kernel module API
[CHIPSEC] OS      : Linux 4.12.12-gentoo #18 SMP Mon Nov 6 21:18:43 UTC 2017 x86_64
[CHIPSEC] Platform: Desktop 6th Generation Core Processor Quad Core (Skylake CPU / Sunrise Point PCH)
[CHIPSEC]      VID: 8086
[CHIPSEC]      DID: 191F
 
[*] loading common modules from "/usr/lib64/python2.7/site-packages/chipsec-1.3.5-py2.7-linux-x86_64.egg/chipsec/modules/common" ..
[+] loaded chipsec.modules.common.spi_desc
[+] loaded chipsec.modules.common.smrr
[+] loaded chipsec.modules.common.spi_fdopss
[+] loaded chipsec.modules.common.smm
[+] loaded chipsec.modules.common.spi_lock
[+] loaded chipsec.modules.common.bios_smi
[+] loaded chipsec.modules.common.bios_kbrd_buffer
[+] loaded chipsec.modules.common.bios_ts
[+] loaded chipsec.modules.common.rtclock
[+] loaded chipsec.modules.common.ia32cfg
[+] loaded chipsec.modules.common.spi_access
[+] loaded chipsec.modules.common.bios_wp
[+] loaded chipsec.modules.common.secureboot.variables
[+] loaded chipsec.modules.common.uefi.access_uefispec
[+] loaded chipsec.modules.common.uefi.s3bootscript
[*] No platform specific modules to load
[*] loading modules from "/usr/lib64/python2.7/site-packages/chipsec-1.3.5-py2.7-linux-x86_64.egg/chipsec/modules" ..
[+] loaded chipsec.modules.remap
[+] loaded chipsec.modules.smm_dma
[+] loaded chipsec.modules.memconfig
[*] running loaded modules ..

[*] running module: chipsec.modules.common.spi_desc
[x][ =======================================================================
[x][ Module: SPI Flash Region Access Control
[x][ =======================================================================
[*] FRAP = 0x00004ACB << SPI Flash Regions Access Permissions Register (SPIBAR + 0x50)
    [00] BRRA             = CB << BIOS Region Read Access 
    [08] BRWA             = 4A << BIOS Region Write Access 
    [16] BMRAG            = 0 << BIOS Master Read Access Grant 
    [24] BMWAG            = 0 << BIOS Master Write Access Grant 
[*] Software access to SPI flash regions: read = 0xCB, write = 0x4A

[+] PASSED: SPI flash permissions prevent SW from writing to flash descriptor

[*] running module: chipsec.modules.common.smrr
[x][ =======================================================================
[x][ Module: CPU SMM Cache Poisoning / System Management Range Registers
[x][ =======================================================================
[+] OK. SMRR range protection is supported

[*] Checking SMRR range base programming..
[*] IA32_SMRR_PHYSBASE = 0xC5400006 << SMRR Base Address MSR (MSR 0x1F2)
    [00] Type             = 6 << SMRR memory type 
    [12] PhysBase         = C5400 << SMRR physical base address 
[*] SMRR range base: 0x00000000C5400000
[*] SMRR range memory type is Writeback (WB)
[+] OK so far. SMRR range base is programmed

[*] Checking SMRR range mask programming..
[*] IA32_SMRR_PHYSMASK = 0xFFC00800 << SMRR Range Mask MSR (MSR 0x1F3)
    [11] Valid            = 1 << SMRR valid 
    [12] PhysMask         = FFC00 << SMRR address range mask 
[*] SMRR range mask: 0x00000000FFC00000
[... 357 lines cut ...]
[CHIPSEC] Modules total         18
[CHIPSEC] Modules failed to run 0:
[CHIPSEC] Modules passed        15:
[+] PASSED: chipsec.modules.common.spi_desc
[+] PASSED: chipsec.modules.common.smrr
[+] PASSED: chipsec.modules.common.spi_fdopss
[+] PASSED: chipsec.modules.common.smm
[+] PASSED: chipsec.modules.common.spi_lock
[+] PASSED: chipsec.modules.common.bios_smi
[+] PASSED: chipsec.modules.common.bios_kbrd_buffer
[+] PASSED: chipsec.modules.common.bios_ts
[+] PASSED: chipsec.modules.common.rtclock
[+] PASSED: chipsec.modules.common.ia32cfg
[+] PASSED: chipsec.modules.common.secureboot.variables
[+] PASSED: chipsec.modules.common.uefi.access_uefispec
[+] PASSED: chipsec.modules.remap
[+] PASSED: chipsec.modules.smm_dma
[+] PASSED: chipsec.modules.memconfig
[CHIPSEC] Modules failed        1:
[-] FAILED: chipsec.modules.common.bios_wp
[CHIPSEC] Modules with warnings 2:
[!] WARNING: chipsec.modules.common.spi_access
[!] WARNING: chipsec.modules.common.uefi.s3bootscript
[CHIPSEC] Modules skipped 0:
[CHIPSEC] ***********************************

Folkerttt, 2017-11-07:
tools.uefi.blacklist and tools.uefi.uefivar_fuzz came out clean. 

Can someone check above log for possible infection?

skochinsky, 2017-11-07:
blacklist checks for only a couple known modules; even if you don't have them doesn't mean you don't have something else. You can try making a whitelist using a known-good update from the vendor and run it against your flash: 
https://github.com/chipsec/chipsec/issues/207#issuecomment-300072485

Folkerttt, 2017-11-07:
I have asus b150m pro gaming motherboard, and the whitelist from chipsec/whitelist.py all seem to fail. 
I can't find the whitelist from Asus for the Asus b150m pro gaming motherboard. 

Also, about the s3bootscript not found, how can i make sure this did not got exploited already?

ErikBjorge, 2017-11-07:
Have you updated to the latest BIOS for your platform?  This may resolve the bios_wp issue.

Have you been able to collect the hashes for your platform using the whitelist tool?  Running `python chipsec_main -m tools.uefi.whitelist -a generate` should create a whitelist for your platform.  The output file should be efilist.json by default.  Just want to confirm that SPI reads and whitelist generation are working correctly.

Also, CHIPSEC failing to find the S3 Boot Script can be a good thing.  This is normally and indicator that the system firmware is protecting the resume script.  Protections may include storing in SMRAM or some other secure location.  Further investigation would be needed to confirm this.

ErikBjorge, 2017-11-07:
It looks like the latest BIOS for the B150M Pro Gaming version is 0601.

You can try to get the .cap file that matches your current firmware version and generate the whitelist file for that version and then check your current firmware image for deltas.

Create a whitelist from .cap file:
`python chipsec_main.py -m tools.uefi.whitelist -a generate,efilist.json,B150M-PRO-GAMING-ASUS-0601.CAP`

Check current firmware against efilist.json:
`python chipsec_main.py -m tools.uefi.whitelist -a check`

ZandroidX, 2018-02-11:
Do not use chipsec it is malware, I ran the rogue mmio tool and it reflashed my firmware, whole system is bricked audio sounds terrible and internet speed is much slower and sometimes hangs. Chipsec is malware, DO NOT USE THIS SOFTWARE*********************************************************************

johnloucaides, 2018-02-12:
If you have concerns about chipsec affecting the system, please update in #344.

ErikBjorge, 2018-05-21:
Closing due to lack of addition information.  Please reopen if you still need support.

closed 2018-05-21 by ErikBjorge