chipsec/chipsec #324: Problems trying to run on windows 10

marcosdiazr, 2017-11-22:
Hi, I'm trying to run chipsec on windows 10.
I think i followed the steps in the manual, i built the driver and copied to the folder, installed setuptools and pywin32, but when /i try to run either chipsec_main or chipsec_util I get the following error:

ERROR: Could not load helper for 'Windows' environment (unsupported environment?)
Traceback (most recent call last):
  File "chipsec_main.py", line 611, in <module>
    sys.exit( main() )
  File "chipsec_main.py", line 607, in main
    chipsecMain = ChipsecMain( argv if argv else sys.argv[1:] )
  File "chipsec_main.py", line 164, in __init__
    self._cs = chipset.cs()
  File "C:\Users\marcos\Documents\chipsec\chipsec\chipset.py", line 788, in cs
    _chipset = Chipset(helper())
  File "C:\Users\marcos\Documents\chipsec\chipsec\helper\oshelper.py", line 475, in helper
    _helper  = OsHelper()
  File "C:\Users\marcos\Documents\chipsec\chipsec\helper\oshelper.py", line 108, in __init__
    raise OsHelperError( "Could not load helper for '%s' environment (unsupported environment?)" % os_system, errno.ENODEV )
chipsec.helper.oshelper.OsHelperError: Could not load helper for 'Windows' environment (unsupported environment?)

BrentHoltsclaw, 2017-11-27:
The error usually means that the helper could not find a driver.  Did you follow steps to sign the driver and put the system into TestSigning mode?

marcosdiazr, 2017-12-12:
Yes I have done that and continue having the same problem. I have put the driver in chipsec\chipsec\helper\win\win7_amd64 is this correct? how can i try to get a more descriptive problem?

ErikBjorge, 2017-12-12:
That is the correct location assuming that this is a 64 bit Windows 10 install.  This exception only happens when the driver file can't be found for your current OS architecture.  Can you also put a version of the driver in chipsec\chipsec\helper\win\win7_x86 to see if it attempts to load it?

marcosdiazr, 2017-12-12:
Sorry about this, the location was ok, the name of the driver was wrong it should be chipsec_hlpr.sys and i had a different name.
Thanks and sorry for the inconvenient!

ErikBjorge, 2017-12-12:
Not a problem at all.  I will leave this issue open for a day or so in case you have any more driver related issues.

closed 2018-01-05 by abazhaniuk