chipsec/chipsec #325: Tianocore page not found (for UEFI shell approach)

jasonbrown1965, 2017-11-29:
On this page:
https://github.com/chipsec/chipsec/wiki/Creating-a-Bootable-DUET-USB-with-UEFI-Shell

"Building a Bootable USB drive with UEFI Shell" contains advice to "Download the UDK from Tianocore" but clicking this link:

https://sourceforge.net/apps/mediawiki/tianocore/index.php?title=UDK2010

... returns a "Whoops, we can't find that page" error.

A search for Tianocore only brings up one result:
https://sourceforge.net/directory/os:windows/?q=Tianocore with a result for EFI Toolkit

... which also leads to a "Whoops, we can't find that page":
https://sourceforge.net/userapps/mediawiki/tianocore/index.php?title=EFI_Toolkit

---

BrentHoltsclaw, 2017-11-29:
It appears that we need to update the documentation. 

UDK2010 is a branch of the edk2 project:
https://github.com/tianocore/edk2/ 

The instructions for DuetPkg can be found at:
https://github.com/tianocore/edk2/tree/master/DuetPkg

ErikBjorge, 2017-11-29:
We should not be using DUET and should actually boot directly to the UEFI Shell.  We will work on updating the instructions to create a UEFI bootable drive with all the required CHIPSEC components.

skochinsky, 2017-11-29:
this post seems to be relevant: http://www.basicinputoutput.com/2016/05/a-tour-of-intel-chipsec.html

jasonbrown1965, 2018-01-16:
Any updates on this? Sorry not much of a tech person.

ErikBjorge, 2018-01-16:
Are you just looking for information on how to run chipsec from the UEFI shell?  Is the system you are wishing to use chipsec on able to boot a UEFI enabled OS or does it only support legacy boot?  I can provide you with instructions on how to create a UEFI boot image for chipsec if this is what you are looking for.

ErikBjorge, 2018-01-16:
To create a bootable UEFI image follow these steps.

1. Format your media as FAT32
2. Create the following directory structure in the root of the new media
    /efi/boot
3. Copy the UEFI shell (Shell.efi) to the /efi/boot directory.  The binary can be found at the following link.
    https://github.com/tianocore/edk2/raw/master/ShellBinPkg/UefiShell/X64/Shell.efi
4. Rename the UEFI shell file to Bootx64.efi
5. Follow the steps in the CHIPSEC manual to copy the remaining required CHIPSEC files to the disk.

You can now use the BIOS setup utility to configure the system to boot off of your new media.  This can normally be done via Boot configuration page.  Your BIOS may also provide a hot key to select a temporary boot target.

Please also note that if UEFI Secure Boot is enabled you will need to disable the feature before booting to the shell on disk will work.

jasonbrown1965, 2018-01-19:
Many thanks @ErikBjorge, that broke it down nicely !

I'm aware that the outputs will not mean much to a non-tech audience (i.e. me), but sometimes even noobz can spot some gaps, as in this thread.

ErikBjorge, 2018-01-19:
Thanks for finding the documentation issue @jasonbrown1965.  These are the types of issues that get missed.  I will get the wiki updated with the new instructions.