chipsec/chipsec #326: Question about Skylake/Kabylake Servier variants

martingalvan, 2017-12-11:
Hi everyone, I have a question regarding the Skylake Server and Kabylake Server architectures. It looks like these Server variants are seen by Chipsec as the same as regular SKL/KBL, and aren't included in the CHIPSET_FAMILY_XEON array:

https://github.com/chipsec/chipsec/blob/master/chipsec/chipset.py#L112

Some modules (e.g. smm_dma) have a check for the current arch not being a Server one. If it is, then the module isn't supported. Therefore, this might be a bug.

Another thing I noticed is that DID 0x1618 (Broadwell Server) is marked as BDW, but in the following line another Broadwell Server is marked as BDX:

https://github.com/chipsec/chipsec/blob/master/chipsec/chipset.py#L180

Could we confirm whether these are bugs?

frinzell, 2017-12-12:
This depends on which Server you are referring to.  Certain server tests expect certain registers or features to be present, not all server types will have them.   The Broadwell server mentioned is an E3 that aligns with the client register definitions and will generally run as such.

abazhaniuk, 2018-01-18:
Hi @frinzell is any E3 servers uses PC (big core) SoC/chipset?

closed 2018-02-18 by abazhaniuk