chipsec/chipsec #327: Spi read address aligned?

marcosdiazr, 2017-12-15:
Hi, I was playing with chipsec and found something:
If i run
`sudo chipsec_util spi read ff8 38 asd.bin`
and then this:
`sudo chipsec_util spi read ff8 40 asd2.bin`
i would expect asd2.bin to be almost the same than asd.bin except 2 bytes bigger
but i see two very different binaries like if it read different addresses
and if i start changing the starting address only like:
`sudo chipsec_util spi read ff6 40 asd2.bin`
 i see the very same dump
is there some alignment requirement for the address needed to dump spi? I have taken a look to ICH manuals and didn't see any alignment requirement or anything like that. Am I missing something?

ErikBjorge, 2017-12-15:
This does look like an alignment issue in the tool so we will need to fix this.  I did some initial testing and it looks like if you keep the flash linear address and size 16 byte aligned the data will be consistent.  I will work to get this fixed in the next release.

Also note that all numeric values in the command are interpreted as hex values.

ZandroidX, 2018-02-11:
DO NOT USE THIS SOFTWARE **************************************
It infects your chipset firmware and will brick your device.

johnloucaides, 2018-02-12:
If you have concerns about chipsec affecting the system outside the SPI address alignment issues, please update in #344.